adelolmo/hd-idle #7: Arm64 - Failure (SCSI response not ok)

V33m, 2019-09-09:
I'm facing an issue which I'm not sure why I'm encountering as everything worked perfectly with armhf (hd-idle v.1.5 though) for the same setup. The only difference is the device (server), so I'm using arm64 instead of armhf.

/etc/default/hd-idle
```
START_HD_IDLE=true
HD_IDLE_OPTS="-i 0 -c ata -s 0 -a /dev/disk/by-id/ata-ST4000DM004-ID -i 600 -a /dev/disk/by-id/ata-ST6000DM003-ID -i 600"
```

systemctl status hd-idle
```
* hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-09-09 23:46:13 CEST; 1h 3min ago
     Docs: man:hd-idle(8)
  Process: 8664 ExecStart=/usr/sbin/hd-idle $HD_IDLE_OPTS (code=exited, status=1/FAILURE)
 Main PID: 8664 (code=exited, status=1/FAILURE)

Sep 09 23:35:03 hostname systemd[1]: Started hd-idle - spin down idle hard disks.
Sep 09 23:35:03 hostname hd-idle[8664]: symlinkPolicy=0, defaultIdle=0, defaultCommand=ata, debug=false, devices=[{sda /dev/disk/by-id/ata-ST4000DM004-ID 600 ata} {sdb /dev/disk/by-id/ata-ST6000DM003-ID 600 ata}], logFile=
Sep 09 23:46:13 hostname hd-idle[8664]: sdb spindown
Sep 09 23:46:13 hostname hd-idle[8664]: 2019/09/09 23:46:13 SCSI response not ok
Sep 09 23:46:13 hostname hd-idle[8664]: SCSI status: 2 host status: 0 driver status: 8
Sep 09 23:46:13 hostname hd-idle[8664]: SENSE:
Sep 09 23:46:13 hostname hd-idle[8664]: 72 00 00 1d 00 00 00 0e 09 0c 00 00 00 00 00 00 00 00 00 00 40 50 00 00 00 00 00 00 00 00 00 00
Sep 09 23:46:13 hostname hd-idle[8664]: NO ADDITIONAL SENSE INFORMATION
Sep 09 23:46:13 hostname systemd[1]: hd-idle.service: Main process exited, code=exited, status=1/FAILURE
Sep 09 23:46:13 hostname systemd[1]: hd-idle.service: Failed with result 'exit-code'
```

Should I skip using ata as default and instead specify for each _-c ata_? It seems that it spindown my sdb drive, but not my first drive (sda). I'll look more into this tomorrow.

V33m, 2019-09-10:
I can confirm that the issue is related to ata, since scsi works fine as seen below.

```
$ sudo systemctl status hd-idle
* hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-09-10 11:55:51 CEST; 3min 32s ago
     Docs: man:hd-idle(8)
 Main PID: 2109 (hd-idle)
    Tasks: 5 (limit: 3350)
   Memory: 4.7M
   CGroup: /system.slice/hd-idle.service
           `-2109 /usr/sbin/hd-idle -i 0 -a /dev/disk/by-id/ata-ST6000DM003-2CY186_WF306G8F -i 180

Sep 10 11:55:51 hostname systemd[1]: Started hd-idle - spin down idle hard disks.
Sep 10 11:55:51 hostname hd-idle[2109]: symlinkPolicy=0, defaultIdle=0, defaultCommand=scsi, debug=false, devices=[{sda /dev/disk/by-id/ata-ST6000DM003-ID 180 scsi}], logFile=
Sep 10 11:59:19 hostname hd-idle[2109]: sda spindown
```

V33m, 2019-09-10:
I was hoping that everything was fine, but it seems that my HDDs spinup without hd-idle noticing and won't get spun down until it recognises that example I play a video from a HDD.

From systemctl status hd-idle
```
Sep 10 18:01:29 hostname hd-idle[4875]: sdb spindown
Sep 10 18:01:30 hostname hd-idle[4875]: sda spindown
```
But the HDDs are actually not standby as seen below.
```
$ sudo smartctl -i -n standby /dev/disk/by-id/ata-ST4000DM004-ID
smartctl 6.6 2017-11-05 r4594  (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in SLEEP mode, exit(2)
```

```
$ sudo smartctl -i -n standby /dev/disk/by-id/ata-ST6000DM003-ID
smartctl 6.6 2017-11-05 r4594  (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
...
...
...
SMART support is: Enabled
Power mode was:   IDLE_B
```

Any advice? Do you think this is related to the fact that I'm using scsi instead of ata which I normally have used?

adelolmo, 2019-09-10:
Hi @V33m 

I agree with you on blaming the `arm64` package. I cross compiled from my `x86_64` desktop to create all the packages (`amd64`, `i386`, `armhf` and `arm64`). Weird that the `arm64` failed but not the `armhf`.
Sadly I don't have any arm64 hardware where I can test. I guess I should start considering getting one.
On what hardware are you running?

V33m, 2019-09-10:
[Khadas VIM3 (Amlogic A311D SBC)](https://www.khadas.com/vim3), however, there are other great arm64 SBCs to choose as example [Raspberry Pi 4 Model B](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/), [ODROID-N2](https://www.hardkernel.com/shop/odroid-n2-with-4gbyte-ram/) etc :)

V33m, 2019-10-24:
> Sadly I don't have any arm64 hardware where I can test. I guess I should start considering getting one.

Have you had any time to look into arm64? If you have done any changes which you think have sorted the issue, let me know so I can give it a try 👍

adelolmo, 2019-10-25:
I got a raspberry 4 but I didn't manage to get a 64bit linux on it yet. The pi is always headless and I have no way to debug when ssh is not built in.
I keep trying every now and then with different options.

The 1.8 release has better error handling, but it won't solve the arm64 issue anyway. So no real need for trying it out.

V33m, 2019-10-25:
Which OSes have you tried? I don't have a RPi4, but you should give the following a try:
- [Balena](https://www.balena.io/blog/balena-releases-first-fully-functional-64-bit-os-for-the-raspberry-pi-4/)
- [Gentoo](https://www.raspberrypi.org/forums/viewtopic.php?f=54&t=249248)
- [Manjaro  (Arch)](https://www.manjaro.org/download/arm/raspebery-pi-4/arm8-raspberry-pi-4-xfce/)
- [Raspbian with 64-bit kernel](https://www.raspberrypi.org/forums/viewtopic.php?t=250730) (32-bit userland)
- [Ubuntu MATE 18.04.2 (Bionic)](https://ubuntu-mate.org/download/)

I've never had any RPis, but no ssh built in sounds odd for me.. Most of my arm SBCs are headless, so built in ssh has always been a requirement to use a OS for me. Let me know when you want to do some testing down the road.. 👍

adelolmo, 2019-10-27:
As an update: I managed to install a arm64 version of ubuntu and I reproduced the error.
Now I "only" have to fix it.

adelolmo, 2019-10-27:
So far I don't have any good news. 
I based the ata sgio command on `hdparm`'s implementation. I ran the command `sudo hdparm -S 10 /dev/sda` to see if it works on arm64 and got the following error:
```
/dev/sda:
 setting standby to 10 (50 seconds)
SG_IO: bad/missing sense data, sb[]:  70 00 01 00 00 00 00 0a 00 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

This is exactly the same error hd-idle throws.

So I will have to re-think (online research) the whole ata sgio command for arm64 :(

V33m, 2019-10-27:
Yeah, we probably have to do some online research to look into this. The positive aspect of hdparm having the same issue is that we can report it and ask them for input. 

A quick google search didn't give me too more, yet this thead could be a start - https://askubuntu.com/questions/768373/hard-drive-error-bad-missing-sense-data

adelolmo, 2020-03-11:
After some digging I ended up in, what it looks, the description of this problem and how to solve it: [SAT-with-UAS-Linux](https://www.smartmontools.org/wiki/SAT-with-UAS-Linux)
It is an interesting reading but the TL;DR is: `ATA` commands might not work if `UAS` is enable.

I was able to solve it following the manual.
@V33m Please let me know if it works for you too.

V33m, 2020-03-11:
Glad to hear that! Did you disable UAS or overwrite SAT detection for the drive? Which arm64 device do you have, Rpi4?

adelolmo, 2020-03-11:
I temporary disable UAS.
```
cat /sys/module/usb_storage/parameters/quirks
2109:0711:u
```
I used Rpi4 with Ubuntu arm64.

V33m, 2020-03-11:
I'm thinking about the same, given the decrease in write/read speed is not an issue for me as none of the drives are SSD. I consider the issue fixed and one should probably mention it in the README

closed 2020-03-11 by V33m

adelolmo, 2020-03-12:
I've added an entry in the readme to [this wiki topic](https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok).