adelolmo/hd-idle #9: removable disk - how to skip

exodus1984, 2019-10-18:
hello,
I have the following problem. I have a memory card reader - it is detected as follows:
```
[    3.184546] sd 6:0:0:0: [sdf] Attached SCSI removable disk
[    3.187044] sd 6:0:0:2: [sdh] Attached SCSI removable disk
[    3.189671] sd 6:0:0:4: [sdj] Attached SCSI removable disk
[    3.193002] sd 6:0:0:1: [sdg] Attached SCSI removable disk
[    3.197299] sd 6:0:0:3: [sdi] Attached SCSI removable disk
```

hd-idle wants to turn off disks in it even though the disks are not there.
```
hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-10-18 10:10:13 CEST; 7min ago
     Docs: man:hd-idle(8)
  Process: 1150 ExecStart=/usr/sbin/hd-idle $HD_IDLE_OPTS (code=exited, status=1/FAILURE)
 Main PID: 1150 (code=exited, status=1/FAILURE)

paź 18 10:00:08 epc systemd[1]: Started hd-idle - spin down idle hard disks.
paź 18 10:00:08 epc hd-idle[1150]: symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, devices=[{sda sda 0 scsi}], logFile=/var/log/hd-idle.log
paź 18 10:10:11 epc hd-idle[1150]: sdd spindown
paź 18 10:10:12 epc hd-idle[1150]: sde spindown
paź 18 10:10:13 epc hd-idle[1150]: sdf spindown
paź 18 10:10:13 epc hd-idle[1150]: 2019/10/18 10:10:13 open /dev/sdf: no medium found
paź 18 10:10:13 epc systemd[1]: hd-idle.service: Main process exited, code=exited, status=1/FAILURE
paź 18 10:10:13 epc systemd[1]: hd-idle.service: Failed with result 'exit-code'.
```

Is there any option to bypass removable drives? Maybe you can do the option bypassing all replaceable disks (e.g. --no-removable)

adelolmo, 2019-10-18:
Hi @exodus1984,

You can avoid this problem by setting the default idle to 0.
I your case, assuming you only want to spin down `sda`, you can use this configuration:
```
HD_IDLE_OPTS="-i 0 -c scsi -a sda -i 600 -l /var/log/hd-idle.log"
```

exodus1984, 2019-10-18:
heh no :)
This is my current configuration
`HD_IDLE_OPTS="-i 0 -a sdb -i 3600 -a sdc -i 3600 -a sdd -i 3600 -a sde -i 3600 -a sda -i 0 -l /var/log/hd-idle.log"`

Unfortunately, now if I add a disk, I must remember to change the configuration. If there was an option to skip removable it would be simpler.

btw 
Can I use sd [bcde]?

adelolmo, 2019-10-18:
Let me analyze your configuration:
```
HD_IDLE_OPTS="-i 0 -a sdb -i 3600 -a sdc -i 3600 -a sdd -i 3600 -a sde -i 3600 -a sda -i 0 -l /var/log/hd-idle.log"
```
`-i 0` indicates no disk is spun down by default
`-a sdb -i 3600` disk `sdb` spins down after 1h
`-a sdc -i 3600` disk `sdc` spins down after 1h
`-a sdd -i 3600` disk `sdd` spins down after 1h
`-a sde -i 3600` disk `sde` spins down after 1h
`-a sda -i 0` disk `sda` never spins down. This piece is not needed because is the default behavior.

But back to your initial log, I noticed this `defaultIdle=600` so that's why i suggested using default idle `0`.

I understand that having to add new disks to the configuration might get annoying and hard to maintain, but it's the only quick solution I can think of right now.

Can you please show the logs of the application running with the new configuration?

exodus1984, 2019-10-18:
```
hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-10-18 15:41:14 CEST; 2s ago
     Docs: man:hd-idle(8)
 Main PID: 20254 (hd-idle)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/hd-idle.service
           └─20254 /usr/sbin/hd-idle -i 0 -a sdb -i 3600 -a sdc -i 3600 -a sdd -i 3600 -a sde -i 3600 -l /var/log/hd-idle.log

paź 18 15:41:14 epc systemd[1]: Started hd-idle - spin down idle hard disks.
paź 18 15:41:14 epc hd-idle[20254]: symlinkPolicy=0, defaultIdle=0, defaultCommand=scsi, debug=false, devices=[{sdb sdb 3600 scsi} {sdc sdc 3600 scsi} {sdd sdd 3600 scsi} {sde sde 3600 scsi}], logFile=/var/log/hd-idle.log
```

> But back to your initial log, I noticed this defaultIdle=600 so that's why i suggested using default idle 0.

I used it in case I added a disc and forgot to add it to sleep after 1 hour. However, -i 0 will be better - I will sooner see that one is not sleeping.

> I understand that having to add new disks to the configuration might get annoying and hard to maintain, but it's the only quick solution I can think of right now.

I understand. My suggestion regarding --no-removable concerns new functionality, for possible implementation in the future.

You have to admit that the current configuration is just ugly :)

V33m, 2019-10-18:
@exodus1984, I advice you to use label, id etc. instead of dev/sda. It's easy and it work perfectly fine if you plug in and unplug drives. In my case, everything work as expected regardless if one drive is connected or three. So in your case, it would be something like:

```HD_IDLE_OPTS="-i 0 -a /dev/disk/by-id/HDD_ID_1 -i 3600 -a /dev/disk/by-id/HDD_ID_2 -i 3600 -a /dev/disk/by-id/HDD_ID_3 -i 3600 -a /dev/disk/by-id/HDD_ID_4 -i 3600"```

or by label if you want:

```HD_IDLE_OPTS="-i 0 -a /dev/disk/by-label/HDD_LABEL_1 -i 3600 -a /dev/disk/by-label/HDD_LABEL_2 -i 3600 -a /dev/disk/by-label/HDD_LABEL_3 -i 3600 -a /dev/disk/by-label/HDD_LABEL_4 -i 3600"```

You can ofcourse use both and mix it. And I don't agree with you that the current configuration is ugly. It's simple and have worked fine for the last ten years

exodus1984, 2019-10-21:
Ok - forget about what I said about configuration. It's ok. I just think it could have been done less lengthy.
However, I think that no disk should not close the program. There should be information about it in the log, but the program should not end with an error. It should work all the time.

```
hd-idle -i 60 -a sdb -i 0 -a sdc -i 0 -a sdd -i 0 -a sde -i 0 -l /var/log/hd-idle.log
symlinkPolicy=0, defaultIdle=60, defaultCommand=scsi, debug=false, devices=[{sdb sdb 0 scsi} {sdc sdc 0 scsi} {sdd sdd 0 scsi} {sde sde 0 scsi}], logFile=/var/log/hd-idle.log
sdf spindown
2019/10/21 09:07:51 open /dev/sdf: no medium found
```
I started the program from the command line to check whether it really comes out with code 1. And unfortunately this is happening - instead of informing that the disk cannot be put to sleep, the program ends its operation.
In my opinion, this is bad program behavior.

btw - the service behaves the same.
hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-10-21 09:01:24 CEST; 7s ago
     Docs: man:hd-idle(8)
  Process: 8387 ExecStart=/usr/sbin/hd-idle $HD_IDLE_OPTS (code=exited, status=1/FAILURE)
 Main PID: 8387 (code=exited, status=1/FAILURE)

```
paź 21 09:00:21 epc systemd[1]: Started hd-idle - spin down idle hard disks.
paź 21 09:00:21 epc hd-idle[8387]: symlinkPolicy=0, defaultIdle=60, defaultCommand=scsi, debug=false, devices=[{sdb sdb 0 scsi} {sdc sdc 0 scsi} {sdd sdd 0 scsi} {sde sde 0 scsi}], logFile=/var/log/hd-idle.log
paź 21 09:01:24 epc hd-idle[8387]: sdf spindown
paź 21 09:01:24 epc hd-idle[8387]: 2019/10/21 09:01:24 open /dev/sdf: no medium found
paź 21 09:01:24 epc systemd[1]: hd-idle.service: Main process exited, code=exited, status=1/FAILURE
paź 21 09:01:24 epc systemd[1]: hd-idle.service: Failed with result 'exit-code'.
```

adelolmo, 2019-10-21:
Hi @exodus1984,

I agree that the service should be more resilient in this case. 
As you suggested, the best thing to do in this case is to log the error and not to kill the program. I will solve this problem in the next release.

Thank you for your explanation :)

closed 2019-10-21 by adelolmo

adelolmo, 2019-10-22:
@V33m @exodus1984 

One thing about using symlinks:
Please make sure that the symlink resolves to a disk and not to a partition.
```
tree /dev/disk/
```
`tree` shows the structure of the devices and resolves the symlinks. Make sure to use only synlinks resolving to e.g. `../../sda` and not to `../../sda1`. 
hd-idle won't apply specific configuration to devices if the symlinks don't resolve to disks. Instead it will use the default configuration.

V33m, 2019-10-22:
Yeah, I know. I see that I should have specified this in my last post. Thanks for the clarification 👍

PR #10 (2019-10-21): Fixes #9 - prevent service to die when a device is not found
Fixes #9 

Log if a device is not accessible and cannot be spun down instead of killing the service.
commit 54d51a73 (2019-10-21): Fixes #9 - prevent service to die when a device is not found
commit fc6f97e6 (2019-10-21): Merge pull request #10 from adelolmo/bug/9-device-not-found

Fixes #9 - prevent service to die when a device is not found