adelolmo/hd-idle #13: Exec format error

V33m, 2020-02-03:
I did a clean install to latest version of Debian Buster arm64 today and my setup work as earlier except from hd-idle which throw the error seen below. I've done two clean installations without luck. The issue might be related to the content in hd-idle.service given the information from systemctl. If I run `sudo /usr/sbin/hd-idle -v -f /etc/default/hd-idle`, I get the following:
```
/usr/sbin/hd-idle: 1: /usr/sbin/hd-idle: Syntax error: word unexpected (expecting ")")
```

systemctl status hd-idle:
```
● hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-02-03 20:13:45 CET; 3min 44s ago
     Docs: man:hd-idle(8)
  Process: 6861 ExecStart=/usr/sbin/hd-idle $HD_IDLE_OPTS (code=exited, status=203/EXEC)
 Main PID: 6861 (code=exited, status=203/EXEC)

Feb 03 21:58:45 Hostname systemd[1]: Started hd-idle - spin down idle hard disks.
Feb 03 21:58:45 Hostname systemd[6861]: hd-idle.service: Failed to execute command: Exec format error
Feb 03 21:58:45 Hostname systemd[6861]: hd-idle.service: Failed at step EXEC spawning /usr/sbin/hd-idle: Exec format error
Feb 03 21:58:45 Hostname systemd[1]: hd-idle.service: Main process exited, code=exited, status=203/EXEC
Feb 03 21:58:45 Hostname systemd[1]: hd-idle.service: Failed with result 'exit-code'.
```

fnordism, 2020-02-04:
I had built and installed hd-Idle on an arm64 / ubuntu / debian derivative (bionic) Raspi 3 hours ago.
If you want you can test the binary package.

[hd-idle_1.8_arm64.zip](https://github.com/adelolmo/hd-idle/files/4150872/hd-idle_1.8_arm64.zip)

fnordism, 2020-02-04:
Could also be that you have to add a SElinux policy if you use hd-idle

adelolmo, 2020-02-04:
My guess is that the binary is running on the wrong architecture.

Can you share the result of: 
`uname -a`
and
`cat /etc/os-release`

What package are you installing? From https://github.com/adelolmo/hd-idle/releases

V33m, 2020-02-04:
That could be it as I'm running the armhf package on arm64 since I faced the following issue when I tried to use the arm64 package of hd-idle some months ago (https://github.com/adelolmo/hd-idle/issues/7). 

@fnordism, thanks, I will try your pre-compiled arm64 package later today and report back 👍

V33m, 2020-02-04:
So, with the arm64 package, the Exec format error is gone. However, the same issue which I've reported earlier, https://github.com/adelolmo/hd-idle/issues/7, is showing. The main difference this time is that the service doesn't fail, but  it keep saying it cannot spindown ata disk /dev/sdb regardless that the disk already has spun down..
```
● hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-04 17:17:05 CET; 3h 13min ago
     Docs: man:hd-idle(8)
 Main PID: 862 (hd-idle)
    Tasks: 8 (limit: 3270)
   Memory: 5.3M
   CGroup: /system.slice/hd-idle.service
           └─862 /usr/sbin/hd-idle -i 0 -c ata -s 1 -l /var/log/hd-idle.log -a /dev/disk/by-id/ID ...

Feb 04 19:51:58 Hostname hd-idle[862]: sda spinup
Feb 04 20:07:08 Hostname hd-idle[862]: sda spindown
Feb 04 20:08:39 Hostname hd-idle[862]: sda spinup
Feb 04 20:16:14 Hostname hd-idle[862]: sdb spindown
Feb 04 20:16:14 Hostname hd-idle[862]: cannot spindown ata disk /dev/sdb
Feb 04 20:16:14 Hostname hd-idle[862]: . SCSI response not ok
Feb 04 20:16:14 Hostname hd-idle[862]: SCSI status: 2 host status: 0 driver status: 8
Feb 04 20:16:14 Hostname hd-idle[862]: SENSE:
Feb 04 20:16:14 Hostname hd-idle[862]: 72 01 00 1d 00 00 00 0e 09 0c 00 00 00 00 00 00 00 00 00 00 40 50 00 00 00 
Feb 04 20:17:45 Hostname hd-idle[862]: NO ADDITIONAL SENSE INFORMATIONsdb spinup
Feb 04 20:23:49 Hostname hd-idle[862]: sda spindown
Feb 04 20:25:20 Hostname hd-idle[862]: sda spinup
Feb 04 20:32:55 Hostname hd-idle[862]: sdb spindown
Feb 04 20:32:55 Hostname hd-idle[862]: cannot spindown ata disk /dev/sdb
Feb 04 20:32:55 Hostname hd-idle[862]: . SCSI response not ok
Feb 04 20:32:55 Hostname hd-idle[862]: SCSI status: 2 host status: 0 driver status: 8
Feb 04 20:32:55 Hostname hd-idle[862]: SENSE:
Feb 04 20:32:55 Hostname hd-idle[862]: 72 01 00 1d 00 00 00 0e 09 0c 00 00 00 00 00 00 00 00 00 00 40 50 00 00 00 0
Feb 04 20:34:26 Hostname hd-idle[862]: NO ADDITIONAL SENSE INFORMATIONsdb spinup
```

V33m, 2020-02-04:
@fnordism, are you using ata or only scsi with hd-idle arm64? Cause I've been using the armhf package for months on arm64 without issues. However, after doing a reinstall with a new kernel, 5.5, the armhf package is suddenly not working. I did another reinstall on kernel 5.3 in mid January on arm64 and there are no issues over there. Strange indeed.

I'm thinking of looking into the Exec format error issue with the armhf package as I think that issue is more solvable than the ata sgio issue with the arm64 package. Any further tips are much appreciated :)

V33m, 2020-02-06:
I'm using the arm64 package and all HDDs spindown when using only scsi, so I don't need to use ata. The only issue left is the spinup/down logging which is nuts as seen below. There are no actual spinup or spindown as I've been listening, but it still loggs so. @fnordism, if you are using the arm64 package, is logging correct for you?

```
● hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-02-06 15:24:31 CET; 1h 49min ago
     Docs: man:hd-idle(8)
 Main PID: 883 (hd-idle)
    Tasks: 13 (limit: 3270)
   Memory: 5.0M
   CGroup: /system.slice/hd-idle.service
           └─883 /usr/sbin/hd-idle -i 0 -s 1 -l /var/log/hd-idle.log -a /dev/disk/by-id/ID ...

Feb 06 15:41:39 Hostname hd-idle[883]: sda spindown
Feb 06 15:43:10 Hostname hd-idle[883]: sda spinup
Feb 06 15:58:20 Hostname hd-idle[883]: sda spindown
Feb 06 15:59:51 Hostname hd-idle[883]: sda spinup
Feb 06 16:10:28 Hostname hd-idle[883]: sdb spindown
Feb 06 16:11:59 Hostname hd-idle[883]: sdb spinup
Feb 06 16:15:01 Hostname hd-idle[883]: sda spindown
Feb 06 16:16:32 Hostname hd-idle[883]: sda spinup
Feb 06 16:31:42 Hostname hd-idle[883]: sda spindown
Feb 06 16:33:13 Hostname hd-idle[883]: sda spinup
Feb 06 16:40:48 Hostname hd-idle[883]: sdb spindown
Feb 06 16:42:19 Hostname hd-idle[883]: sdb spinup
Feb 06 16:48:23 Hostname hd-idle[883]: sda spindown
Feb 06 16:49:54 Hostname hd-idle[883]: sda spinup
Feb 06 17:05:04 Hostname hd-idle[883]: sda spindown
Feb 06 17:06:35 Hostname hd-idle[883]: sda spinup
Feb 06 17:11:08 Hostname hd-idle[883]: sdb spindown
Feb 06 17:12:39 Hostname hd-idle[883]: sdb spinup
```

adelolmo, 2020-02-07:
@V33m Do the `sda` and `sdb` spindown after they were active?

That log looks like `sda` spins down after 15 mins of inactivity, but spins up right afterwards. Same for `sdb`, after 30 mins it spins down.
I would say that some program is writing/reading both HDs, because that's the only thing that makes any sense to me.

V33m, 2020-02-07:
Yeah, both drives spindown after 15 mins. But they do not spin up again even if hd-idle says so, as both drives make noise when spinning, so it's simple to verify if the drives are spinning or not. So, the arm64 package has apparently two issues, where the first is ata sgio using `-c ata` and the second is wrongful logging.

I would have used the armhf package I've been using for the last months if it hadn't been for the `Exec format error` which I've faced on latest mainline kernel and Armbian with Debian. I've switched back to my earlier setup using hdparm and legacy hd-idle 1.0.5 on my arm64 server. Will retry the armhf package in some weeks to see if the `Exec format error` is gone 👍

V33m, 2020-02-19:
I did a new clean install earlier today and the _Exec format error_ is completely gone... I think it was related to some packages relevant for using armhf on arm64.

The second issue that happened at the same time is the logging which I'll open a separate issue for in the next couple of days after doing some testing. The odd thing regarding one of my HDD's is the suddenly change of id from ata-ST6000DM003-.. when using kernel 5.3 and lower to usb-Seagate_Expansion_Desk_.. for kernel 5.4 and higher. So, I think the issue is related to that change and how usb-management is working for the recent kernels. I'll probably test longtime Kernel 4.9 earlier this week to see if that is working too.

closed 2020-02-19 by V33m

adelolmo, 2020-02-19:
Thank you for the info.

V33m, 2020-02-29:
I've been using hd-idle 1.8 (armhf) on kernel 4.9 (arm64) for several days and everything works perfectly fine! So, it seems that there have been some changes to the USB driver(s) for mainline kernel since the logging of spin ups/downs are completely off seen in the previous message where none of the drives actually did spin up or down. Given that I've no need of using mainline, I'm fine at longtime kernel 4.9 and there are no need to investigate this further for the time beeing👍