adelolmo/hd-idle #17: No log file despite -l /path/to/file.log

agcmartins, 2020-03-31:
Hi,
Fist congratulation for the project I'm really enjoying the program and it works for me. I'm trying to use the log file to track how many times the disk spin up and down during a normal day. I need this to find the best sleep time to balance between disk durability/performance, by reducing the number of spinning up events per day and not allowing the disk to stay up 24/7. I'm using a raspberry 4 and a external HD as a backup centre for my other computers.
I've changed the last line of /etc/default/hd-idle to:
HD_IDLE_OPTS="-i 600 -l /home/USER/hd-idle.log"
I've also enabled and started it in systemctl and I can see that it is working and includes the log option using htop. However after I watched the disk spin up and down for a while, there is no log file in the directory.
Best

adelolmo, 2020-04-01:
Hi @agcmartins 

Could you please share the result of the command `systemctl status hd-idle`?
Also I'd be great if you could double check that the directory `/home/USER` (I guess obfuscated) really exists.

Thank you

agcmartins, 2020-04-01:
Hi,
Directory exist, is my home folder, but I've change it to USER to post.
systemctl status hd-idle
● hd-idle.service - hd-idle - spin down idle hard disks
   Loaded: loaded (/lib/systemd/system/hd-idle.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-03-31 23:47:30 -03; 11h ago
     Docs: man:hd-idle(8)
 Main PID: 371 (hd-idle)
    Tasks: 10 (limit: 4915)
   Memory: 4.2M
   CGroup: /system.slice/hd-idle.service
           └─371 /usr/sbin/hd-idle -i 600 -l /home/USER/hd-idle.log

Mar 31 23:47:30 raspberrypi systemd[1]: Started hd-idle - spin down idle hard disks.
Mar 31 23:47:30 raspberrypi hd-idle[371]: symlinkPolicy=0, defaultIdle=600, defaultCommand=sc
Mar 31 23:59:57 raspberrypi hd-idle[371]: sda spindown
Apr 01 11:07:53 raspberrypi hd-idle[371]: sda spinup

adelolmo, 2020-04-01:
Sadly the log doesn't show long lines :/
Could you run this command and post the result? `grep hd-idle /var/log/syslog`

agcmartins, 2020-04-01:
Sure, but sorry, before I saw you message I had to reboot it.
I've also installed hd-idle in my desktop to test, but leave the log as default (/var/log/hd-idle.log), than after while I have checked the folder and there is no file there too.

Apr  1 11:07:53 raspberrypi hd-idle[371]: sda spinup
Apr  1 11:18:03 raspberrypi hd-idle[371]: sda spindown
Apr  1 11:57:42 raspberrypi hd-idle[371]: sda spinup
Apr  1 12:01:31 raspberrypi systemd[1]: Started hd-idle - spin down idle hard disks.
Apr  1 12:01:32 raspberrypi hd-idle[391]: symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=/home/USER/hd-idle.log, devices=
Apr  1 12:23:05 raspberrypi hd-idle[391]: sda spindown

adelolmo, 2020-04-01:
I can't tell why your log file is always empty. But not all is lost for you.

You can use `grep hd-idle /var/log/syslog` to monitor spinups and spindowns. That will give you the stats you need.

This is a sample of how the log file looks like:
```
date: 2020-04-01, time: 00:00:34, disk: sda, running: 671, stopped: 42523
date: 2020-04-01, time: 00:00:34, disk: sdb, running: 609, stopped: 9638
date: 2020-04-01, time: 00:00:34, disk: sdc, running: 611, stopped: 9576
date: 2020-04-01, time: 12:00:25, disk: sdb, running: 671, stopped: 42519
date: 2020-04-01, time: 12:01:26, disk: sdc, running: 671, stopped: 42581
```
`hd-idle` creates entries in the log file only on spinups. So I don't think it will be any useful for your use case to use the log file.

Hope this helps.

agcmartins, 2020-04-01:
Thank you very much, I'll use that information. Do you have any advice about spin up events and hardware durability for external usb HD?
Regarding the log file, I've notice, that if I create an empty file using log name in directory, the program start to write lines on it. But it seems that if the file doesn't exit the program do not create it.
Best,

adelolmo, 2020-04-01:
I had to re-test the scenario that you describe and I can confirm that the log file _does_ get created if it doesn't previously exits.

Sorry that I can't help you further with this topic, but at least you got the data you were looking for.
Thank you

closed 2020-04-01 by adelolmo

giig1967g, 2022-01-18:
Hi all,
I know it's an old thread, but I also cannot see the log file despite the configuration file has been setup properly.
This is the result of: `grep hd-idle /var/log/syslog`

```
Jan 18 18:55:07 OPENMEDIAVAULT systemd[1]: Started hd-idle - spin down idle hard disks.
Jan 18 18:55:07 OPENMEDIAVAULT hd-idle[936]: symlinkPolicy=0, defaultIdle=0, defaultCommand=scsi, debug=false, logFile=/var/log/hd-idle.log, devices={name=sdf, givenName=/dev/disk/by-label/media1, idle=3600, commandType=scsi}{name=sdh, givenName=/dev/disk/by-label/media2, idle=3600, commandType=scsi}{name=sde, givenName=/dev/disk/by-label/media3, idle=3600, commandType=scsi}{name=sdb, givenName=/dev/disk/by-label/media4, idle=3600, commandType=scsi}{name=sdg, givenName=/dev/disk/by-label/backup1, idle=3600, commandType=scsi}{name=sdd, givenName=/dev/disk/by-label/downloads, idle=3600, commandType=scsi}{name=sdc, givenName=/dev/disk/by-label/rootdata2, idle=3600, commandType=scsi}

```

no log file:
```
user@OPENMEDIAVAULT:~$ cat /var/log/hd-idle.log
cat: /var/log/hd-idle.log: No such file or directory

```

Anything I can do to help you track down the issue?

rabelux, 2022-01-18:
Have you seen my reply to #63?

giig1967g, 2022-01-18:
All right.
I understand now.
Thanks for your quick reply.