adelolmo/hd-idle #18: PPA or other repository for Ubuntu

nickcmaynard, 2020-06-04:
Hello!  Have you considered providing a PPA for this package?  I find it very helpful on my machine, but keeping it up to date is a bit difficult... a PPA would make it super easy.

adelolmo, 2020-06-04:
So far I didn't considered any option to distribute it other than manually downloading the packages.

I certainly understand your point of view. I personally had to figure this one out myself some time ago and that's why I created a personal debian repo to manage my packages.
If you feel comfortable with it, you can find instructions [here](https://adelolmo.github.io/) on how to set it up. If your distro is not one of the existing ones (`bionic`, `xenial` or `trusty`) I suggest you just take `bionic`.

I will leave this issue open though. Having a PPA at some point sounds reasonable for the long term.

V33m, 2020-06-05:
I've a small issue with your personal repository as I can't find any of the packages there, seen from below:
```
$ wget -O- http://adelolmo.github.io/andoni.delolmo@gmail.com.gpg.key | sudo apt-key add -
$ echo "deb http://adelolmo.github.io/$(lsb_release -cs) $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/adelolmo.github.io.list
$ sudo apt update
...
Hit https://adelolmo.github.io/buster buster InRelease
$ sudo apt-cache policy hd-idle
hd-idle:
  Installed: (none)
  Candidate: 1.05+ds-2~bpo10+1
  Version table:
     1.05+ds-2 650
        650 http://deb.debian.org/debian testing/main arm64 Packages
     1.05+ds-2~bpo10+1 680
        680 http://httpredir.debian.org/debian buster-backports/main arm64 Packages
        680 http://deb.debian.org/debian buster-backports/main arm64 Packages
$ ls /var/lib/apt/lists/
adelolmo.github.io_buster_dists_buster_InRelease
adelolmo.github.io_buster_dists_buster_main_binary-armhf_Packages.lz4
apt.armbian.com_dists_buster_buster-desktop_binary-arm64_Packages.lz4
...
```
Any ideas?

nickcmaynard, 2020-06-05:
Thanks for considering this @adelolmo.  I won't be taking your personal repo, as I'm not after the other packages, but it's useful to have the option!

adelolmo, 2020-06-05:
@V33m 
I just added the package for the architecture `arm64` to `buster`, `xenial` and `trusty` distros.
I hope that solves your problem.

V33m, 2020-08-14:
I actually can't find the arm64 package for buster or at your [PACKAGES.html](https://adelolmo.github.io/PACKAGES.html). I can see the armhf package with `sudo apt policy hd-idle:armhf`. Is it possible to add Focal Fossa to the distribution given it's LTS and supported until 2025? Thanks in advance 👍

adelolmo, 2020-08-15:
In my previous comment I wrote `buster` instead of `bionic` :-/ 
This is how it looks like now:
| Distribution | Component | Architecture | Package | Version |
| ------------ | ------ | -------- | ------- | ------- |
|focal|main|amd64|hd-idle|1.10|
|focal|main|arm64|hd-idle|1.10|
|bionic|main|amd64|hd-idle|1.10|
|bionic|main|arm64|hd-idle|1.10|
|xenial|main|amd64|hd-idle|1.10|
|xenial|main|arm64|hd-idle|1.10|
|trusty|main|amd64|hd-idle|1.10|
|trusty|main|arm64|hd-idle|1.10|
|buster|main|armhf|hd-idle|1.10|
|buster|main|arm64|hd-idle|1.10|
|stretch|main|armhf|hd-idle|1.10|
|stretch|main|arm64|hd-idle|1.10|
|jessie|main|armhf|hd-idle|1.10|
|jessie|main|arm64|hd-idle|1.10|

V33m, 2020-08-15:
The arm64 version is showing for buster now. And thanks by adding focal as well!

adelolmo, 2020-08-16:
I've added armhf to focal too.

closed 2020-12-06 by adelolmo

auanasgheps, 2020-12-06:
Hey, why there's no amd64 repo for buster?

`N: Skipping acquire of configured file 'main/binary-amd64/Packages' as repository 'http://adelolmo.github.io/buster buster InRelease' doesn't support architecture 'amd64'`

adelolmo, 2020-12-06:
> Hey, why there's no amd64 repo for buster?
> 
> `N: Skipping acquire of configured file 'main/binary-amd64/Packages' as repository 'http://adelolmo.github.io/buster buster InRelease' doesn't support architecture 'amd64'`

I never needed it before.
Now is added. Please try again.

auanasgheps, 2020-12-06:
> > Hey, why there's no amd64 repo for buster?
> > `N: Skipping acquire of configured file 'main/binary-amd64/Packages' as repository 'http://adelolmo.github.io/buster buster InRelease' doesn't support architecture 'amd64'`
> 
> I never needed it before.
> Now is added. Please try again.

It does! Thanks, now I can keep my hd-idle updated.