adelolmo/hd-idle #19: hd-idle does not detect drive spinning up due to `smartctl` activity

nickcmaynard, 2020-06-05:
When drives are spun "up" by smartctl activity (such as a `smartctl -x /dev/sdX` run), hd-idle does not detect this activity.

It appears that its internal state remains at "spunDown=true", and consequently the drive is never spun back down again.

Is there any way that this situation can be accounted for?

V33m, 2020-06-05:
Adelolmo probably has a better answer, yet I don't think it's possible given how hd-idle detects if a disk is spinning or not. Regardsless, this is a corner case as one does not very often make a smartmoontool report. In addition to that, when you actually use the disk next time afterwards, the internal state will switch and hd-idle will detect when the disk can be spun down. So, the detection issue is only relevant for the period between when the disk is spun up with smartmoontool -x and before you read/write to the disk next time

adelolmo, 2020-06-05:
@V33m explained exactly how hd-idle behaves in this case.
hd-idle monitors `/proc/diskstats` in order to evaluate disk activity. If smartmontools spins up the disk without modifying reads/writes in `/proc/diskstats` hd-idle won't notice that change. Therefore hd-idle will only spin down the disk after a change in `/proc/diskstats`.
I don't see how to workaround this problem.

nickcmaynard, 2020-06-05:
Thank you both - that's a good explanation.

I wonder - would there be overhead/consequences from periodically (say, every configurable X seconds) reissuing the standby command if a disk "should be" spun down according to hd-idle's internal state?

It's not perfect and arguably not particularly elegant, but it would compensate for this edge case.

nickcmaynard, 2020-06-05:
I apologise - I think this may be influenced by a misunderstanding of the "command type" parameter.  The disk in question is a USB WD drive.  Though it logs no errors with `scsi` as the command type, it appears that doesn't actually work... and it *appeared* that smartctl was spinning things up (it wasn't).  With `ata`, errors are logged, but it does actually work!

I think this is a PEBKAC issue.  Perhaps some extended guidance on when to select `ata` vs `scsi` would help?

nickcmaynard, 2020-06-05:
<deleted a comment that muddied the water - created #20 instead>

V33m, 2020-06-06:
Spinning up/down a hard drive is quite hardware tear problematic when done frequent, and every time you reissue a standby command you either spin up or down the disk, regardless of the disk's previous state. I would highly advice against that. I think I would rather fork smartmoontool, build your own version where you change the code of smartmoontool to do a simple seek at root of the drive when making the report. That would fix this issue as smartmontool will change /proc/diskstats.

Out of curiosity, why do you need to make smartmoontool reports so often? Isn't it enough with one report per harddrive and that's it?

closed 2020-08-04 by adelolmo

marki555, 2021-01-18:
You have 2 options regarding smartmontools:
1) When you run smartctl from cron for example, include any action which would increase the counters (list directory, use `dd`, ...)
2) If using smartd to continously monitor the disk state, use smartctl's option "-n STANDBY" which will prevent smartctl/smartd spinning up the disk

graemev, 2022-11-18:
For most folks **hdparm -C** would do the job. Sadly no use to me (on QNAP) since there it's compiled with debug on so it writes to syslog and hence to disk (spinning it up)

devZer0, 2023-03-10:
When it's cause by smartd, you can use

/dev/disk/by-id/<devicename> -d ignore
before
DEVICESCAN statement
in smartd.conf
to exclude the disk from being queried by smartd

see
https://www.smartmontools.org/ticket/1390#comment:2