adelolmo/hd-idle #20: Extend guidance on `ata` vs `scsi` command types

nickcmaynard, 2020-06-05:
I have misunderstood the "command type" parameter.

For some USB drives (including those with `/dev/disk/by-id` entries starting, interestingly, `ata-`) the default `scsi` command does not work.

I've seen this with Seagate USB drives such as the Seagate Backup Plus Hub, which looks like `/dev/disk/by-id/ata-ST6000DM003-2CY186-...`.

Though when hd-idle logs no errors with `scsi` as the command type, it appears that doesn't actually work.  With `ata`, errors are logged, but it does actually work!

Perhaps some extended guidance on when to select `ata` vs `scsi` would help?  I'll take a look and see if I can proffer a PR.

adelolmo, 2020-06-05:
I have a briefly introduction to the topic in the [readme](https://github.com/adelolmo/hd-idle#background).

This is my current setup:

```
tree /dev/disk/
...
├── by-id
│   ├── ata-WDC_*** -> ../../sdb
│   ├── ata-WDC_***-part1 -> ../../sdb1
...
│   ├── usb-Intenso_External_USB_3.0_*** -> ../../sda
│   ├── usb-Intenso_External_USB_3.0_***-part1 -> ../../sda1
│   ├── usb-WD_Elements_*** -> ../../sdc
│   ├── usb-WD_Elements_***-part1 -> ../../sdc1
│   ├── usb-WD_Elements_*** -> ../../sdd
│   └── usb-WD_Elements_***-part1 -> ../../sdd1
├── by-label
│   ├── boot -> ../../mmcblk0p1
│   ├── disk1 -> ../../sdb1
│   ├── disk3 -> ../../sda1
│   ├── parity1 -> ../../sdd1
│   └── rootfs -> ../../mmcblk0p2
├── by-partlabel
│   ├── disk2 -> ../../sdc1
│   └── Elements -> ../../sdd1
...
```
| drive | name | description | command type |
|-----|-----|-----|-----|
|sda | disk3 | usb-Intenso_External_USB_3.0_*** | scsi |
|sdb | disk1 | ata-WDC_*** | ata |
|sdc | disk2 | usb-WD_Elements_*** | scsi |
|sdd | parity1 | usb-WD_Elements_*** | scsi |

Once you have play around with your disk you can bring your results as PR if you want it.

adelolmo, 2021-11-25:
> Though when hd-idle logs no errors with scsi as the command type, it appears that doesn't actually work. With ata, errors are logged, but it does actually work!

This should be fix by #58

closed 2023-05-07 by adelolmo