adelolmo/hd-idle #29: the disk won't spin up!

maxlinux2000, 2020-08-11:
Hi
I have installed from you .deb release hd-idle and it can spin down the disk after 10 minuts.
the problem is thar the disk can't spin up when I need to read something on the disk.

The disk export a folder via nfs for other computers on my home network, but I use it via vnc too as local files.

My system is a RaspberryPi4 (32bit)  an the disk is a normal usb3 3.5" of 2 TB with his own power supply

Do you need some others infos? How to obtain them?

adelolmo, 2020-08-11:
Hi @maxlinux2000 
You just described the same setup I currently have running. It works like a charm for me.

First of all clarify that hd-idle doesn't actually spin up disks. That it's handled by the system.

Could you share your `/var/log/hd-idle.log`? And the result of `grep hd-idle /var/log/syslog`?

closed 2020-08-19 by adelolmo

dmr138, 2022-10-22:
I am having a similar issue although the result of `grep hd-idle /var/log/syslog` is just `grep: /var/log/syslog: binary file matches`. drive seems to spin down fine but then are inaccessible through the file system when trying to access later i get `ls: cannot access '*dirname*': Input/output `error`` .

any ideas?