adelolmo/hd-idle #30: errors building from source in a Raspberry Pi 4B

hydra3333, 2020-08-18:
Hello.  Thanks for this utility.

I have a Raspberry Pi 4 ... so I tried to build hd-idle from source per instructions at https://github.com/adelolmo/hd-idle#build-from-source and received errors as shown below.

Suggestions would be appreciated. 

Thanks.

# Install go:
```
cd ~/Desktop
wget https://golang.org/dl/go1.15.linux-armv6l.tar.gz
sudo tar -C /usr/local -xzf go1.15.linux-armv6l.tar.gz
chmod +777 /usr/local/go
chmod +777 -R /usr/local/go
```
# added these 2 lines to /etc/profile :
```
sudo nano /etc/profile
export PATH=$PATH:/usr/local/go/bin
export GOPATH=/usr/local/go
```

#Rebooted.

#Ran the go test at https://golang.org/doc/install section "Test your installation" which was succfessful.

#Per instructions at https://github.com/adelolmo/hd-idle#build-from-source :
```
mkdir -p $GOPATH/src/github.com/adelolmo
cd $GOPATH/src/github.com/adelolmo
git clone https://github.com/adelolmo/hd-idle.git
cd hd-idle
dpkg-buildpackage -a armhf -us -uc -b
```

# Resulting log with errors copied from the console:

```
dpkg-buildpackage: info: source package hd-idle
dpkg-buildpackage: info: source version 1.10
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Andoni del Olmo <andoni.delolmo@gmail.com>
dpkg-buildpackage: info: host architecture armhf
 dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9) golang-go:native (>= 1.3.3) dh-golang
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
```

hydra3333, 2020-08-18:
OK, so I did 

`sudo apt -y install debhelper`

and re-did `dpkg-buildpackage -a armhf -us -uc -b`   
and now the console log shows:
```
dpkg-buildpackage: info: source package hd-idle
dpkg-buildpackage: info: source version 1.10
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Andoni del Olmo <andoni.delolmo@gmail.com>
dpkg-buildpackage: info: host architecture armhf
 dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: golang-go:native (>= 1.3.3) dh-golang
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
```

being a newbie with `go`, I am unsure where to go next (pun intentional))

hydra3333, 2020-08-18:
OK, 
tried `sudo apt -y install  dh-make-golang` and `dh-make-golang` which did install something,
Then tried `dpkg-buildpackage -a armhf -us -uc -b` again, and it seemed to build.

Cheers

V33m, 2020-08-18:
Is there a reason for why you want to build it by yourself when adelolmo provides already pre-built packages? https://github.com/adelolmo/hd-idle/releases

hydra3333, 2020-08-18:
Yes, with some software I occasionally take an interest to look at the  code and build it myself.

adelolmo, 2020-08-18:
`dh_golang errors out because the go compiler used was not installed via a package.  Therefore the step is skipped`

This is just a [comment](https://github.com/adelolmo/hd-idle/blob/3e9692a8823cb7f3ab1abe95b44901ec9575313f/debian/rules#L72) in the build process.

It is meant to allow to build if you don't install go via [package](https://packages.debian.org/stretch/golang-go), which is exactly my setup. I prefer to run on the latest go version and don't wait for the package.

adelolmo, 2020-08-18:
BTW the package is created under `/usr/local/go/src/github.com/adelolmo` in your filesystem.

hydra3333, 2020-08-19:
Cool, thanks.

closed 2020-08-19 by hydra3333