adelolmo/hd-idle #32: 64bit version for new raspberry pi o/s ?

hydra3333, 2020-10-17:
hello, I see Raspberry Pi is "soon" to release a 64-bit operating system which is very close to vanilla debian.

the beta testing thread for that is here https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275372

I am testing a few things on the 64bit beta o/s now, but am unsure what it means for hd-idle and whether a 64-bit build is required.

are you able to clarify, and perhaps add a 64bit release for the Raspberry Pi o/s ?

V33m, 2020-10-17:
Arm 64-bit is already provided with [hd-idle_1.*_arm64.deb](https://github.com/adelolmo/hd-idle/releases/latest). Furthermore, you are able to use armhf packages as well, so you can use [hd-idle_1.*_armhf.deb](https://github.com/adelolmo/hd-idle/releases/latest) instead of the 64-bit if you have any issues. For instance ["SCSI response not ok"](https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok) which might occur for the arm64 packge but not the armhf.

adelolmo, 2020-10-18:
As @V33m already mentioned, arm64 packages are available.
You can also build the package yourself running:
```
dpkg-buildpackage -a arm64 -us -uc -b
```

closed 2020-10-18 by adelolmo

hydra3333, 2020-10-19:
Ah.  Thank you.
So
```
git clone https://github.com/adelolmo/hd-idle.git
cd hd-idle
#dpkg-buildpackage -a arm64 -us -uc -b
dpkg-buildpackage -a armhf -us -uc -b
dpkg -i ../hd-idle*.deb
```
and ignore any error messages.

adelolmo, 2020-10-19:
> and ignore any error messages.

Please don't ignore _error_ messages.
Given that you managed to build the package in the [past](https://github.com/adelolmo/hd-idle/issues/30), I don't expect you'll have any problem now.