adelolmo/hd-idle #36: Feature proposal: Add an option to query disk spinning state

jk-1, 2020-12-04:
Proposal is to add a new option to `hd-idle` for reporting whether a disk is spinning or not.

Example (a

    hd-idle -a sdb -q 

Output e.g.

    /dev/sdb spinning

Example (b

    hd-idle -a sdc -q 

Output e.g.

    /dev/sdc idle

Example (c

    hd-idle -q 

Output e.g.

    /dev/sdb spinning
    /dev/sdc idle

I have asked a solution for this also on [superuser,](https://superuser.com/questions/1595505/how-to-detect-if-usb3-hd-is-spinning) but so far without any success.

I can spin down my WD MyBook USB3 HD boxes with `hd-idle`, so most likely, it should be possible to add a reporting option to the program.

Thank you for your help.

adelolmo, 2020-12-04:
Hi @jk-1 

Checking for disk activity state is completely out of the scope of this tool.

I'd suggest to take a look to [SCSI response not ok](https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok) as an starting point to fix the issue with `sudo hdparm -C /dev/sdb`.

If you are already using `hd-idle`, you can always check disks' spin changes in system log:
```
$ grep hd-idle /var/log/syslog
Dec  4 00:00:34 enterprise hd-idle[30189]: sda spinup
Dec  4 00:00:34 enterprise hd-idle[30189]: sdc spinup
Dec  4 00:00:34 enterprise hd-idle[30189]: sdb spinup
Dec  4 00:15:34 enterprise hd-idle[30189]: /dev/sdb spindown
Dec  4 00:16:35 enterprise hd-idle[30189]: /dev/sda spindown
Dec  4 00:16:35 enterprise hd-idle[30189]: /dev/sdc spindown
Dec  4 09:30:39 enterprise hd-idle[30189]: sda spinup
Dec  4 09:31:39 enterprise hd-idle[30189]: sdc spinup
Dec  4 09:41:39 enterprise hd-idle[30189]: /dev/sda spindown
Dec  4 09:41:39 enterprise hd-idle[30189]: /dev/sdc spindown
Dec  4 12:00:42 enterprise hd-idle[30189]: sda spinup
Dec  4 12:00:42 enterprise hd-idle[30189]: sdc spinup
Dec  4 12:16:42 enterprise hd-idle[30189]: /dev/sda spindown
Dec  4 12:16:42 enterprise hd-idle[30189]: /dev/sdc spindown
Dec  4 13:06:45 enterprise hd-idle[30189]: sdc spinup
Dec  4 13:16:45 enterprise hd-idle[30189]: /dev/sdc spindown
Dec  4 13:22:48 enterprise hd-idle[30189]: sda spinup
Dec  4 13:22:48 enterprise hd-idle[30189]: sdc spinup
Dec  4 13:22:48 enterprise hd-idle[30189]: sdb spinup
Dec  4 13:33:48 enterprise hd-idle[30189]: /dev/sda spindown
Dec  4 13:33:48 enterprise hd-idle[30189]: /dev/sdc spindown
Dec  4 13:33:51 enterprise hd-idle[30189]: /dev/sdb spindown
```
You can know the state of any disk by checking the latest state change in the log. In the example above `sda`, `sdb` and `sdc` are all spun down.

cannondale0815, 2020-12-05:
A good, non-intrusive way to query the spindown state is to use smartctl:

`sudo smartctl -i -n standby /dev/sdc | grep "Power" | awk '{print $4}'`

The above either returns "ACTIVE" or nothing. The latter means the drive is spun down.

auanasgheps, 2020-12-05:
> Hi @jk-1
>
> Checking for disk activity state is completely out of the scope of this tool.
>
> I'd suggest to take a look to [SCSI response not [...]

Why saved logs only show when the disk are spun up and not also when they're down like the command above? Would be possible to get this?

jk-1, 2020-12-05:
> A good, non-intrusive way to query the spindown state with smartctl is as follows:
> 
> `sudo smartctl -i -n standby /dev/sdc | grep "ACTIVE\|STANDBY" | awk '{print $4}'`
> 
> The above either returns "ACTIVE" or "STANDBY".
> 
> The "-n" option is essentially preventing the drive from spinning back up if it's in standby mode. The built-in help says:
> 
> ` -n MODE[,STATUS], --nocheck=MODE[,STATUS] No check if: never, sleep, standby, idle (see man page)`

For some reason this does not work. I have tried several modes but the disk spins up, before the result is displayed.
E.g

```
sudo smartctl -i -n sleep,0 /dev/sdb
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.0-56-generic] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in SLEEP mode, exit(0)
```

Before the last row is displayed, disk is spinning again.

jk-1, 2020-12-05:
> Checking for disk activity state is completely out of the scope of this tool.
> 
> I'd suggest to take a look to [SCSI response not ok](https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok) as an starting point to fix the issue with `sudo hdparm -C /dev/sdb`.
> 

Hi adelolmo,

Thank you for the tip: https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok

I tried that:

    lsusb
    Bus 002 Device 003: ID 1058:25ee Western Digital Technologies, Inc. My Book 25EE
    ...
    
and then 

    cat /sys/module/usb_storage/parameters/quirks
    1058:25ee:u

Then plug out and plug in the disk. Unfortunately hdparm still cannot get the disk state.

    sudo hdparm -C /dev/sdb1
    
    /dev/sdb1:
    SG_IO: bad/missing sense data, sb[]:  f0 00 01 00 50 00 81 0a 80 00 00 00 00 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
     drive state is:  unknown

adelolmo, 2020-12-06:
> > Hi @jk-1
> > Checking for disk activity state is completely out of the scope of this tool.
> > I'd suggest to take a look to [SCSI response not ok](https://github.com/adelolmo/hd-idle/wiki/SCSI-response-not-ok) as an starting point to fix the issue with `sudo hdparm -C /dev/sdb`.
> > If you are already using `hd-idle`, you can always check disks' spin changes in system log:
> > ```
> > $ grep hd-idle /var/log/syslog
> > ```
> 
> Why saved logs only show when disks are spun up and not also when they're down like the command above? Would be possible to get this?

Please check the documentation [to understand how the logs](https://github.com/adelolmo/hd-idle#understand-the-logs).

adelolmo, 2020-12-06:
@jk-1 

I won't be able to give you any support on tools like `hdparm` or `smartctl`. 
You should try asking on their project sites: https://sourceforge.net/projects/hdparm/ and https://github.com/mirror/smartmontools

I'm closing this issue, but feel free to keep the conversation active if you need to.

closed 2020-12-06 by adelolmo