adelolmo/hd-idle #37: hd-idle not starting on Ubuntu 20.4.01

cannondale0815, 2020-12-05:
I tried installing the latest 1.11 amd64 package on Ubuntu 20.04.1 LTS. No matter which command line option I choose, even when I just use -h, it only prints the following and then hangs:

`symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=`

I previously had the original hd-idle from C. Mueller installed on my system (built from source). Before installing this new version of hd-idle, I made sure to remove the old binary to ensure there are no conflicts, but unsure if there are any other residual files that may cause issues on my end.

I also tried building this version of hd-idle from source and downloaded and installed the latest 1.15.6 version of the "go" environment beforehand. It's in my path and I can call "go" manually without issues. But when I try to build hd-idle as instructed in the documentation, I am nevertheless getting the following unmet dependency issue:

`dpkg-checkbuilddeps: error: Unmet build dependencies: golang-go:native (>= 1.3.3) dh-golang`

I would love to use this version of hd-idle on my Ubuntu system, but am stuck. Any help is appreciated :)

adelolmo, 2020-12-05:
Hi @cannondale0815 

I'm not sure what you mean by "hangs". `hd-idle` runs in the background.

Could you please share the results of the following commands?
- `ls /dev/s*`
- `tree /dev/disk/`
- `cat /etc/default/hd-idle`

cannondale0815, 2020-12-05:
@adelolmo by hangs, I mean it outputs that one line I mentioned above and doesn't return to the command prompt.

Here is the output of the commands you requested:

**1)**
`user@NAS:~$ ls /dev/s*
/dev/sda   /dev/sdb2  /dev/sde   /dev/sdg1  /dev/sdj   /dev/sdl1  /dev/sdo   /dev/sdq1  /dev/sg12  /dev/sg2  /dev/sg7       /dev/stdin
/dev/sda1  /dev/sdc   /dev/sde1  /dev/sdh   /dev/sdj1  /dev/sdm   /dev/sdo1  /dev/sg0   /dev/sg13  /dev/sg3  /dev/sg8       /dev/stdout
/dev/sda2  /dev/sdc1  /dev/sdf   /dev/sdh1  /dev/sdk   /dev/sdm1  /dev/sdp   /dev/sg1   /dev/sg14  /dev/sg4  /dev/sg9
/dev/sdb   /dev/sdd   /dev/sdf1  /dev/sdi   /dev/sdk1  /dev/sdn   /dev/sdp1  /dev/sg10  /dev/sg15  /dev/sg5  /dev/snapshot
/dev/sdb1  /dev/sdd1  /dev/sdg   /dev/sdi1  /dev/sdl   /dev/sdn1  /dev/sdq   /dev/sg11  /dev/sg16  /dev/sg6  /dev/stderr

/dev/serial:
by-id  by-path

/dev/shm:
sem.9334581e-7251-4ef7-a8ec-5bfe8e89ff68  sem.F8EEB26D-AB23-4061-8250-130AB5285A4A

/dev/snd:
by-path  controlC0  hwC0D3  pcmC0D3p  pcmC0D7p  seq  timer`

**2)**
'user@NAS:~$ tree /dev/disk
/dev/disk
├── by-id
│   ├── ata-SAMSUNG_MZNLF128HCHP-00000_S28TNXAH209310 -> ../../sdn
│   ├── ata-SAMSUNG_MZNLF128HCHP-00000_S28TNXAH209310-part1 -> ../../sdn1
│   ├── ata-Samsung_SSD_840_EVO_250GB_S1DBNSAF354840P -> ../../sdb
│   ├── ata-Samsung_SSD_840_EVO_250GB_S1DBNSAF354840P-part1 -> ../../sdb1
│   ├── ata-Samsung_SSD_840_EVO_250GB_S1DBNSAF354840P-part2 -> ../../sdb2
│   ├── ata-Samsung_SSD_850_EVO_250GB_S3PZNF0JB05079T -> ../../sda
│   ├── ata-Samsung_SSD_850_EVO_250GB_S3PZNF0JB05079T-part1 -> ../../sda1
│   ├── ata-Samsung_SSD_850_EVO_250GB_S3PZNF0JB05079T-part2 -> ../../sda2
│   ├── ata-ST4000VN000-1H4168_Z300P70F -> ../../sdc
│   ├── ata-ST4000VN000-1H4168_Z300P70F-part1 -> ../../sdc1
│   ├── ata-ST4000VN000-1H4168_Z300TE47 -> ../../sdj
│   ├── ata-ST4000VN000-1H4168_Z300TE47-part1 -> ../../sdj1
│   ├── ata-ST4000VN000-1H4168_Z30135GV -> ../../sdi
│   ├── ata-ST4000VN000-1H4168_Z30135GV-part1 -> ../../sdi1
│   ├── ata-ST4000VN000-1H4168_Z301395T -> ../../sdk
│   ├── ata-ST4000VN000-1H4168_Z301395T-part1 -> ../../sdk1
│   ├── ata-ST4000VN000-1H4168_Z30139FG -> ../../sdh
│   ├── ata-ST4000VN000-1H4168_Z30139FG-part1 -> ../../sdh1
│   ├── ata-ST4000VN000-1H4168_Z30147EP -> ../../sdg
│   ├── ata-ST4000VN000-1H4168_Z30147EP-part1 -> ../../sdg1
│   ├── ata-ST4000VN008-2DR166_WDH1B545 -> ../../sdq
│   ├── ata-ST4000VN008-2DR166_WDH1B545-part1 -> ../../sdq1
│   ├── ata-TOSHIBA_HDWQ140_579IK0C3FPBE -> ../../sdl
│   ├── ata-TOSHIBA_HDWQ140_579IK0C3FPBE-part1 -> ../../sdl1
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E2XTY3KS -> ../../sde
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E2XTY3KS-part1 -> ../../sde1
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E3KZD3P1 -> ../../sdm
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E3KZD3P1-part1 -> ../../sdm1
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E3KZD5ZE -> ../../sdf
│   ├── ata-WDC_WD40EFRX-68WT0N0_WD-WCC4E3KZD5ZE-part1 -> ../../sdf1
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HK2SSBN -> ../../sdo
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HK2SSBN-part1 -> ../../sdo1
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HK8WGVN -> ../../sdd
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HK8WGVN-part1 -> ../../sdd1
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HKAGDGN -> ../../sdp
│   ├── ata-WDC_WD80EMAZ-00WJTA0_7HKAGDGN-part1 -> ../../sdp1
│   ├── md-name-NAS:0 -> ../../md0
│   ├── md-uuid-b93d3ef7:3a319f4f:e5f862b9:4d980338 -> ../../md0
│   ├── wwn-0x50000397cbf809a9 -> ../../sdl
│   ├── wwn-0x50000397cbf809a9-part1 -> ../../sdl1
│   ├── wwn-0x5000c500641aa140 -> ../../sdc
│   ├── wwn-0x5000c500641aa140-part1 -> ../../sdc1
│   ├── wwn-0x5000c50064aaa8c2 -> ../../sdj
│   ├── wwn-0x5000c50064aaa8c2-part1 -> ../../sdj1
│   ├── wwn-0x5000c50065404633 -> ../../sdi
[... 136 lines cut ...]
#                          which are not named otherwise by using this
#                          parameter. This can also be a symlink
#                          (e.g. /dev/disk/by-uuid/...)
#  -i <idle_time>          Idle time in seconds.
#  -c <command_type>       Api call to stop the device. Possible values are "scsi"
#                          (default value) and "ata".
#  -s symlink_policy       Set the policy to resolve symlinks for devices.
#                          If set to "0", symlinks are resolve only on start.
#                          If set to "1", symlinks are also resolved on runtime
#                          until success. By default symlinks are only resolve on start.
#                          If the symlink doesn't resolve to a device, the default
#                          configuration will be applied.
#  -l <logfile>            Name of logfile (written only after a disk has spun
#                          up). Please note that this option might cause the
#                          disk which holds the logfile to spin up just because
#                          another disk had some activity. This option should
#                          not be used on systems with more than one disk
#                          except for tuning purposes. On single-disk systems,
#                          this option should not cause any additional spinups.
#
# Options not exactly useful here:
#  -t <disk>               Spin-down the specified disk immediately and exit.
#  -d                      Debug mode. It will print debugging info to
#                          stdout/stderr (/var/log/syslog if started as with systemctl)
#  -h                      Print usage information.`

adelolmo, 2020-12-05:
@cannondale0815 

>  by hangs, I mean it outputs that one line I mentioned above and doesn't return to the command prompt

I suggest you have a look to the section [Run hd-idle](https://github.com/adelolmo/hd-idle/blob/master/README.md#run-hd-idle) in the README. Specially the part that mentions _If the Debian package was installed_.

Also, your configuration in `/etc/default/hd-idle` is missing the entry `HD_IDLE_OPTS`. Please check [the configuration section](https://github.com/adelolmo/hd-idle/blob/master/README.md#configuration). This is the original `/etc/default/hd-idle` [content](https://github.com/adelolmo/hd-idle/blob/master/debian/hd-idle.default) in case you need it.

cannondale0815, 2020-12-05:
> @cannondale0815
> I suggest you have a look to the section [Run hd-idle](https://github.com/adelolmo/hd-idle/blob/master/README.md#run-hd-idle) in the README. Specially the part that mentions _If the Debian package was installed_.
> 
> Also, your configuration in `/etc/default/hd-idle` is missing the entry `HD_IDLE_OPTS`. Please check [the configuration section](https://github.com/adelolmo/hd-idle/blob/master/README.md#configuration). This is the original `/etc/default/hd-idle` [content](https://github.com/adelolmo/hd-idle/blob/master/debian/hd-idle.default) in case you need it.

@adelolmo For now, I am just trying to run it on the command line, not as a service, And when I just say `hd-idle -h`, it only shows:

```
user@NAS:~$ sudo /usr/sbin/hd-idle -h
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=
```

And doesn't return (I have to Ctrl+C to get back to prompt). Am I misunderstanding how this version of hd-idle has changed from the original one? That one, I could just invoke from the command line.

adelolmo, 2020-12-05:
> Am I misunderstanding how this version of hd-idle has changed from the original one? That one, I could just invoke from the command line.

It depends on what's you expectation here. What are you trying to achieve?

`hd-idle` works like a service: it's started and it runs until stopped.
The only usecase that returns to the prompt is when invoked with the flag `-t` to spin down a single disk.

As a side note: I noticed a bug invoking help via `-h` flag. I'll fix it in the next release.

cannondale0815, 2020-12-05:
Ah -- I guess that may explain it. Since I have been using the old hd-idle for a few years, I was stuck thinking I could just invoke your version the same way on the command line. And the first thing I always do is to have a program print it's own help screen to familiarize myself with it. So let me go ahead and configure it to run as a service and if I run into any issues, I'll report back -- thank you!

cannondale0815, 2020-12-06:
I set it up to run as a service with systemctl and it works great, so thank you for the explanations above, I appreciate it!

closed 2020-12-06 by adelolmo