adelolmo/hd-idle #38: only calculate partition access

teknoid, 2020-12-30:
Hi Adelolmo,

I have installed monitorix to watch my system, incl. disk temp and smart reporting. The data collection script runs every minute but this increase the diskstats counters for the disk devices (sda, sdb, etc...)

Thats why hd-idle never spins down the devices even if there is no disk activity on it's partitions.

Could you please add a config switch that ignores the device entry in /proc/diskstats and counts only the partition entries sda1...sdaX? Or make this the (IMHO) default behavior and add a config switch to include the device entry.

Thanks and best regards
Heiko

eloo, 2021-03-07:
Hi,
just found that issue and i can confirm that checking only the disk will cause disk from spinning down.
In my case smartctl -n standy seems also to change the counts for the disks.

But i'm curious what will happen if we configure in hd-idle sda1 instead of sda?

adelolmo, 2021-03-12:
@eloo 
Currently you have to provide [devices and not partitions](https://github.com/adelolmo/hd-idle/blob/master/diskstats/snapshot.go#L90).

eloo, 2021-03-15:
@adelolmo 
okay.. do you think it would be possible to use a partition there? so to adjust the regex there?

i've quickly tested/checked the stats file for the partition and it looks like it could work without any big changes

mindrunner, 2021-03-22:
It seems like this is an issue only in more recent kernels. Similar problem here:

https://github.com/desbma/hddfancontrol/issues/28

Unfortunately, there is currently no way to have disk monitoring set up (temps, smart, power state) and spin down drives when Idle. (Workaround: Use 5.4 LTS)

The `partition` thing sounds like a good idea. I would love to see that, too

teknoid, 2021-03-23:
Indeed I was thinking that it worked in the past. But I did not recognize that kernel update might be the reason.

closed 2021-03-28 by adelolmo

mindrunner, 2021-04-06:
Checking back in to confirm that this indeed solves my problem. Disks spinning down again with more recent kernels!

eloo, 2021-04-06:
I can also confirm that its now working as expected. 
Great work.

PR #48 (2021-03-28): Use partitions read/write to calculate disk activity
- Fixes #38 

It changes the method to calculate disk activity. Now the disk activity is calculated by watching read/write changes on partition level instead of disk level.