adelolmo/hd-idle #39: Crash with some command line arguments

mtklr, 2021-01-19:
Hi,

Certain combinations of arguments (or too many?) seem to cause a crash (i'm using raspbian lite):

```
$ hd-idle -ad
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=
^C
$ hd-idle -a -d
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=true, logFile=, devices={name=-d, givenName=-d, idle=600, commandType=scsi}
^C
$ hd-idle -da
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=
^C
$ hd-idle -d -a
panic: runtime error: index out of range [3] with length 3

goroutine 1 [running]:
main.main()
	/media/andoni/two/projects/go/src/github.com/adelolmo/hd-idle/main.go:81 +0xd5c

```

Thanks for your time,
Matt

JabelsBS, 2021-01-31:
I have the same Problem, can't start the service over hd-idle.
`pi@raspberrypi:~ $ hd-idle
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=
^C`

mtklr, 2021-01-31:
> I have the same Problem, can't start the service over hd-idle.
> 
> `pi@raspberrypi:~ $ hd-idle`
> 
> `symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, debug=false, logFile=, devices=`
> 
> `^C`
> 
> I installed the latest .deb 

Hi,

Just to be clear; in my example above, the first three commands successfully start hd-idle, it's the last one with the "panic" response that is the problem. :)

JabelsBS, 2021-01-31:
> Hi,
>
> Just to be clear; in my example above, the first three commands successfully start hd-idle, it's the last one with the "panic" response that is the problem. :)

Sorry i misunderstood that

PR #42 (2021-02-07): Prevent crash on missing arguments
- Fixes #39
closed 2021-02-07 by adelolmo