adelolmo/hd-idle #40: getting error during building

mirek190, 2021-01-31:
...
github.com/adelolmo/hd-idle/io
github.com/adelolmo/hd-idle/vendor/github.com/benmcclelland/sgio
github.com/adelolmo/hd-idle/sgio
github.com/adelolmo/hd-idle
# github.com/adelolmo/hd-idle
src/github.com/adelolmo/hd-idle/hdidle.go:145: undefined: math.RoundToEven
dh_auto_build: go install -v -p 1 github.com/adelolmo/hd-idle github.com/adelolmo/hd-idle/diskstats github.com/adelolmo/hd-idle/io github.com/adelolmo/hd-idle/            sgio github.com/adelolmo/hd-idle/vendor/github.com/benmcclelland/sgio returned exit code 2
debian/rules:51: recipe for target 'override_dh_auto_build' failed
make[1]: *** [override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/mirek190/hd-idle'
debian/rules:43: recipe for target 'build' failed
make: *** [build] Error 2

adelolmo, 2021-02-01:
Are you following[ this steps](https://github.com/adelolmo/hd-idle#build-from-source) to build?

mirek190, 2021-02-01:
Yes I flowed steps.
Even used go 1.15+

by this command

dpkg-buildpackage -a armhf -us -uc -b

adelolmo, 2021-02-02:
Hi @mirek190 

In the first comment I see that you build for the architecture `amd64`, but in the second comment you use `armhf`. Is your target architecture different than the building one? Otherwise, you can just run `dpkg-buildpackage -us -uc -b`.

mirek190, 2021-02-02:
I only used what is in the manual ! 
Copy - paste ... did not noticed there is for arm ... lol

dpkg-buildpackage -a armhf -us -uc -b

why there is armhf ?    ;-D

closed 2021-02-03 by adelolmo