adelolmo/hd-idle #44: [Feature Request] Integrate logrotate configuration

auanasgheps, 2021-03-05:
Hi,
I think would be great if you could integrate the logrotate configuration for hd-idle, so when a user enables logs does not get flooded by them.

I don't know where to place the PR, so here's the config I'm using:
`/etc/logrotate.d/hd-idle`
```
/var/log/hd-idle.log {
      missingok
      notifempty
      compress
      delaycompress
}
```

adelolmo, 2021-03-07:
This idea was already in my radar. I left it out of the last release, but I'm certainly will add it in the next one.

My internal debate was the following:
* Since the log file is present in the default configuration, hd-idle should be responsible of rotate it.
* Adding log rotation might brake installations where users do some statistics from the full log file.

So I still have to decide how best introduce rotation. (weekly, monthly, yearly, per size, etc)

auanasgheps, 2021-03-07:
What you mean as "default" configuration? The example in the main page?

I cant' remember where the config I posted comes from, but archives logs every week and keeps 4 weeks. It's pretty standard and works really well.

adelolmo, 2021-03-08:
> What you mean as "default" configuration? The example in the main page?

[This is the default](https://github.com/adelolmo/hd-idle/blob/master/debian/hd-idle.default#L36) log file.

I'll try your example (which is pretty similar to the [original hd-idle](https://salsa.debian.org/debian/hd-idle/-/blob/master/debian/hd-idle.logrotate) rotation) and will definitely consider adding it on the next release.

auanasgheps, 2021-03-08:
I guess is the same file, probably is where I've taken the config file but I could not remember, makes sense now :)

PR #47 (2021-03-21): Add logrotate
- Fixes #44 

Add a one week log rotation for `/var/log/hd-idle.log`.
closed 2021-03-21 by adelolmo