adelolmo/hd-idle #45: Hard drive doesn't spin down after certain uptime

bkev, 2021-03-18:
I am running hd-idle on a Pi v4 and it works really well. What I have noticed though is that if my pi hasn't been rebooted for a while (3 weeks for example) the hard drive doesn't spin down anymore. If I restart the hd-idle service it works for a short period again but then doesn't spin down again. 
Are there any logs I can look at to see why this is happening and if it is hd-idle? I appreciate it could be something else causing it too. 
Thanks.

adelolmo, 2021-03-18:
Hi @bkev 
You can start by checking [the logs documentation](https://github.com/adelolmo/hd-idle#understand-the-logs).

bkev, 2021-03-18:
Cheers I'll keep an eye on it. There aren't as many entries when it stops working but I'll see what happens now as I've just upgraded to the latest version and rebooted.

closed 2021-03-28 by adelolmo