adelolmo/hd-idle #46: Docker image

tekgator, 2021-03-20:
Hi Andoni,
not really an issue but couldn't find another way to contact.

I have created a Docker image for your fabular piece of software. It runs very well on my Homelab to spin down the HDD's. 
Maybe you can promote it in your Readme to get more user utilizing it.

The repository is: https://github.com/tekgator/docker-hd-idle
Docker Hub: https://hub.docker.com/r/tekgator/docker-hd-idle

Patrick

adelolmo, 2021-03-21:
Hi @tekgator,

There is already another docker image out there as far as I know: https://hotio.dev/containers/hdidle
I don't feel comfortable promoting one or the other since I'm not a user of any of them, and I can't judge its quality.
Would you mind explaining me what's the goal on having a container running hd-idle? What prevents you from using the debian package directly?

tekgator, 2021-03-21:
Oh I didn't even know :)... That's why I was creating my own image... Completly understand the your strugle to decide. Not a problem.

The reason for docker? I assume the easyness for getting it up and running and furthermore monitoring it. I have integrated my docker container monitoring into Home Assistant.

But also I personally know a lot of people which are not into Linux at all but know how to start a container via portainer.

adelolmo, 2021-03-21:
Thank you for your explanation.
It's always interesting how other folks handle things.
I'll close the issue now.

closed 2021-03-21 by adelolmo

Skaronator, 2023-06-01:
The [hotio](https://hotio.dev/containers/hdidle) image is gone and @tekgator image is still on 1.02 while 1.20 is out. Would be cool to have an offical image hosted on the GitHub container registry.

adelolmo, 2023-06-01:
> an official image hosted on the GitHub container registry

Packaging software to docker is out of my comfort zone. 
Me doing it won't make it more "official" than anyone else's image. If there's a real a need for it, I would expect some community to come up with its own container and giving it some sort of support.

alex3305, 2025-08-22:
For those encountering this issue through a search engine. You can easily run a replacement.

## Docker Compose

The following will download and install hd-idle every time the container is started.

```yaml
services:
  hd-idle:
    image: alpine
    container_name: hd-idle
    privileged: true
    restart: unless-stopped
    command: >
      sh -c "apk add --no-cache hd-idle &&
             hd-idle -i 900 -a sda -i 600"
```

## Dockerfile

Or you can build your image:

```dockerfile
FROM alpine
RUN apk add --no-cache hd-idle
ENTRYPOINT ['/usr/sbin/hd-idle']
```

Build with `docker buildx build -t hd-idle .` and run with `docker run --name hd-idle --privileged hd-idle -i 900 -a sda -i 600`.

> [!IMPORTANT]
> These are very basic examples which should work out of the box.