adelolmo/hd-idle #50: no diskstats with encrypted hard drive   1.14 / 2021-03-28

stivi2000, 2021-04-02:
Hello my english is not that good.

The latest update does not detect any activity on encrypted hard drives (LUKS).

best greets Stefan

adelolmo, 2021-04-03:
First of all, I'm going to need some info from your system.
* `lsblk`
* `cat /proc/diskstats`
* `cat /etc/default/hd-idle`

Then:
1. Add `-d` to `HD_IDLE_OPTS`. Documentation is [here](https://github.com/adelolmo/hd-idle#Configuration).
2. Restart the service: `systemctl restart hd-idle`.
3. Leave it running for as long as it needs for the disk to stop.
4. Run `grep hd-idle /var/log/syslog`

stivi2000, 2021-04-03:
Hello with this version (1.13 / 2021-03-04) everything works.
Is currently installed.

lsblk
sda           8:0    0 111.8G  0 disk
├─sda1        8:1    0   256M  0 part  /boot
└─sda2        8:2    0 111.6G  0 part  /
sdb           8:16   0   5.5T  0 disk
└─sdb-crypt 254:0    0   5.5T  0 crypt /sftp/Stefan/6TB_Filme
sdc           8:32   0   5.5T  0 disk
└─sdc-crypt 254:1    0   5.5T  0 crypt /srv/dev-disk-by-uuid-9ef70e95-d07b-4650-a5a1-2b99e7446a28
sdd           8:48   0 931.5G  0 disk
└─sdd-crypt 254:2    0 931.5G  0 crypt /srv/dev-disk-by-uuid-7da00225-261d-4af2-94ca-4e1652b5a986
sde           8:64   1 114.6G  0 disk
└─sde1        8:65   1 114.6G  0 part  /sftp/Stefan/StickPI

   8       0 sda 25563 20954 2128848 38074 22493 20778 1075906 71784 0 126690 113231 0 0 0 0 6106 3371
   8       1 sda1 410 4017 65964 739 2 0 2 0 0 1200 740 0 0 0 0 0 0
   8       2 sda2 25044 16937 2048458 37212 22413 20778 1075904 71767 0 125980 108979 0 0 0 0 0 0
   8      16 sdb 3473070 561 911729241 2436589 1481 1035 17992 15375 0 9530810 2465631 0 0 0 0 535 13666
   8      32 sdc 2891 192 57743 13523 1085550 14035 982686648 9819204 0 5102050 10209416 0 0 0 0 12140 376689
   8      48 sdd 972 144 42638 3498 1 0 8 1 0 5250 3500 0 0 0 0 0 0
   8      64 sde 370 67 42218 984 14 12 138 52 0 1560 1036 0 0 0 0 0 0
   8      65 sde1 286 67 29842 756 14 12 138 52 0 1450 808 0 0 0 0 0 0
 254       0 dm-0 3473100 0 911713802 13983560 2516 0 20128 18530 2 9980310 14002090 0 0 0 0 0 0
 254       1 dm-1 2446 0 41658 36560 622848 0 982735200 122960440 0 6763530 122997000 0 0 0 0 0 0
 254       2 dm-2 524 0 26746 3220 1 0 8 0 0 2230 3220 0 0 0 0 0 0

# defaults file for hd-idle

# start hd-idle automatically?
START_HD_IDLE=true

# hd-idle command line options
# Options are:
#  -a <name>               Set device name of disks for subsequent idle-time
#                          parameters (-i). This parameter is optional in the
#                          sense that there's a default entry for all disks
#                          which are not named otherwise by using this
#                          parameter. This can also be a symlink
#                          (e.g. /dev/disk/by-uuid/...)
#  -i <idle_time>          Idle time in seconds.
#  -c <command_type>       Api call to stop the device. Possible values are "scsi"
#                          (default value) and "ata".
#  -s symlink_policy       Set the policy to resolve symlinks for devices.
#                          If set to "0", symlinks are resolve only on start.
#                          If set to "1", symlinks are also resolved on runtime
#                          until success. By default symlinks are only resolve on start.
#                          If the symlink doesn't resolve to a device, the default
#                          configuration will be applied.
#  -l <logfile>            Name of logfile (written only after a disk has spun
#                          up). Please note that this option might cause the
#                          disk which holds the logfile to spin up just because
#                          another disk had some activity. This option should
#                          not be used on systems with more than one disk
#                          except for tuning purposes. On single-disk systems,
#                          this option should not cause any additional spinups.
#
# Options not exactly useful here:
#  -t <disk>               Spin-down the specified disk immediately and exit.
#  -d                      Debug mode. It will print debugging info to
#                          stdout/stderr (/var/log/syslog if started as with systemctl)
#  -h                      Print usage information.
HD_IDLE_OPTS="-i 0 -c ata -d -l /var/log/hd-idle.log -a /dev/disk/by-id/ata-ST6000DM003-2CY186_ZR10E9G3 -i 1800 -a /dev/disk/by-id/ata-WDC_WD60EZRZ-00GZ5B1_WD-WX51D88D23P2 -i 1800"

adelolmo, 2021-04-04:
Thank you for reporting this.
I'll handle this scenario in the next release.

stivi2000, 2021-04-04:
okay thank you

the update works with an encrypted partition

just not if the hard drive is encrypted

best greets Stefan

dieneuser, 2021-04-05:
> okay thank you
> 
> the update works with an encrypted partition
> 
> just not if the hard drive is encrypted
> 
> best greets Stefan

Hey stivi/Stefan,

I can confirm the behaviour. I have exactly the same error pattern with encrypted hard drives. With version 1.13 everything workes.

PR #51 (2021-04-05): Handle disks with no partitions
- Fixes #50 

Encrypted disks do not have any partitions. In this case, the disk level activity has to be taken into consideration.
closed 2021-04-05 by adelolmo