adelolmo/hd-idle #55: Guidance on spinups vs idle?

bitwombat, 2021-06-07:
Hi Andoni, thanks for hd-idle.

If I spin down my disk, it's 88% of the time, which seems like a win for device longevity.

But, it spins up 10 times per day, increasing wear-and-tear.

Do you have any idea what the balance is between those two? Does the 88% idle time beat out the spin-ups for increasing the life of the disk, or the other way around?

adelolmo, 2021-11-25:
Sorry, I don't have any insights about what's the impact of the life of the disks.

closed 2021-11-25 by adelolmo

rabelux, 2021-11-26:
@bitwombat have a look at the datasheet. There you should find specs for `Load/Unload Cycles` and `Mean Time Between Failures (MTBF, hours)`.
A [Seagate IronWolf](https://www.seagate.com/files/www-content/datasheets/pdfs/ironwolf-16tb-DS1904-19-2110US-en_US.pdf) for example should be able to spin up 600.000 times in total, which means you can spin up 600 times a day with a life expectancy of 3 years. A big margin for safety is always recommended as these are average values.

bitwombat, 2021-11-26:
Good advice, very helpful @rabelux Thanks!