adelolmo/hd-idle #59: Idle time problem

Schneck1, 2021-09-08:
I compiled latest version on a new Debian 11 bullseye system and have an issue with the idle-time.
HD spins not down when i'm using time greater then 500.
tried several variants of time - working fine till value 500. 
With Debian 10 buster its working - same setup.

/etc/default/hd-idle:
`START_HD_IDLE=true
HD_IDLE_OPTS="-i 0 -a sdb -i 1200 -l /var/log/hd-idle.log"`

Schneck1, 2021-09-08:
seems like its a kernel problem; with older buster kernel 4.19.0-17 its ok...

adelolmo, 2021-09-10:
Hi @Schneck1 ,
My guess is that after 500 seconds, the new kernel produces some activity on the disk.
Could you maybe enable the debug mode and share here the results? 
To do so, just add the flag _-d_: `HD_IDLE_OPTS="-i 0 -a sdb -i 1200 -l /var/log/hd-idle.log" -d`

Schneck1, 2021-09-10:
you're right, as you can see the kernel causes every 10mins a lastIO entry...

[https://paste.kodi.tv/ubepuvowoc](url)

adelolmo, 2021-09-11:
Looking at the logs on the 5.10.0-8-amd64 kernel I see reads every 10 minutes. I can't imagine what's causing the reads.
Could you please have a look to https://github.com/adelolmo/hd-idle/issues/38?

closed 2021-11-18 by adelolmo