adelolmo/hd-idle #63: No output in log file

JuergenBausa, 2021-11-02:
I just installed current hd-idle (deb of 1.16) from github on my armbian buster system (odroid hc2). I use
the systemd unit with

`START_HD_IDLE=true
HD_IDLE_OPTS="-a sda -i 1800 -l /var/log/hd-idle.log"`

in /etc/default/hd-idle (its actually two separate lines, I dont know what goes wrong with the quoting here).

Spinning down seems to work (at least from the messages in /var/log/syslog, I have no physical access
to the system at the moment), but I dont find any output in the log file /var/log/hd-idle.log.

What is the problem?

Jürgen

rabelux, 2021-11-03:
I had the same "issue" as I thought the logfile should get written when a disk goes to sleep. It doesn't. You have to wake up a disk to produce an output to the logfile.
An entry at 9:00 for example saying the disk was running for 2 hrs and stopped for 1 hr means the disk was woken up at 6, stopped at 8 and was now woken up again. Hope that helps.

JuergenBausa, 2021-11-03:
Thanks, for the hint. You are right. I just had to wait, now I see something in the log.

closed 2021-11-03 by JuergenBausa

adelolmo, 2021-11-03:
FYI: https://github.com/adelolmo/hd-idle#understand-the-logs