adelolmo/hd-idle #65: /dev/disk/by-id in addition or instead to /dev/sdX in the logfile.

gert99, 2021-11-24:
This would make it possible to get long term statistics from the logfile, because /dev/sdX could have be changed after a reboot.

adelolmo, 2021-11-25:
Out of curiosity, what are you trying to achieve?

gert99, 2021-11-26:
I wrote a simple bash script that reads the entire logfile (plus the rotated ones) and outputs the following values from it for each disk:

1) How many days are recorded in the log files.

2) For each hard disk: How many spin-up operations
	Today, Yesterday, Total, Total/number of days
	
3) For each hard disk: The spin-up duration in hours
	Today, Yesterday, Total, Total/Number of days
	
I want to use this to get an overview of the individual usage of each hard drive. Unfortunately, this no longer works if the assignment of /dev/sdX to the individual devices has changed.

PR #94 (2023-03-05): Use GivenName instead of Name in the log
Change the device name in log and stdout to GivenName instead of Name.
Adding or removing disks will affect the assignment of /dev/sdX. So it makes sense to set a unique id for each disk. (This is common, just as [scrutiny](https://github.com/AnalogJ/scrutiny) also changed sdX to disk id).
Obviously, users with such a purpose won't pass sdX as a parameter, as they are dynamic. So I think using GivenName directly is acceptable. Also, for those users who do pass sdX as a parameter, the original functionality will not change.

Closes #65
closed 2023-03-05 by adelolmo