adelolmo/hd-idle #67: Option to pick between disk vs partition activity.

KocsisV, 2021-12-25:
I recently updated from 1.13 to 1.16 and I ran into an issue:

I have disks which are passed through to a virtual machine, thus the partitions never get accessed by the host. The /proc/diskstats never reports any activity for the partitions, only for the disk. All disks (try to) spin down when the set amount of time elapses after boot, even when there is activity. And never after.
Would it be possible to add a flag/option to select between the disk based vs partition based monitoring methods? (And add a note/warning for now that it may falsely mark the drive as active if some monitoring tool accesses it.)

Before marking the drive as "spun up" it would be possible to check it's power state. hdparm/smartctl supports getting the power state (idle/active, standby) without waking up the drive. diskstats's metrics increases when I check the power state of a drive (the drive itself does not spin up) but hd-idle (1.13) logs its as a spinup event.

Thanks for this useful tool!

adelolmo, 2022-07-30:
Hi @KocsisV 
Would you mind to try out version [1.17](https://github.com/adelolmo/hd-idle/releases/tag/v1.17) an see if it makes any difference?

devZer0, 2022-12-16:
does hd-idle monitor only partition activity to decide to shutdown a disk or not? 

i'm using a disk as a whole device in qemu and hd-idle at the host level does not seem to see that there is io being active. 

from what the numbers tell, it only seems to look at the partitions disk stats. 

can you provide an option for looking at the device disk stats or tell why it's only looking at the partition disk stats ?

devZer0, 2022-12-16:
would you provide an option to monitor at the disk level?  i have passed trough the disk which i like to spin down to virtual machine, and partition counters do not update that way

from inside the virtual machine i cannot spin down the disk, apparently the command will not pass trough

closed 2023-05-07 by adelolmo