adelolmo/hd-idle #68: hd-idle killed the disk?

m33x, 2022-01-14:
I used hd-idle to spin down my USB HDD on my Raspberry Pi 4. But even after restarting, and on other PCs (running macOS and Windows 10 which do not have hd-idle installed) after connecting the drive it does nothing. It never starts, it never spins up again. It gets power, the white LED is on, but nothing else. Is it dead Jim?

Disk: Western Digital Elements 4TB (WDBHDW0040BBK) USB 3.0 Micro-B

adelolmo, 2022-01-14:
> Is it dead Jim?
Given the results from your investigation, I'd say it is dead.
I won't be able to guess how or why, though.

m33x, 2022-01-14:
Thanks. I ordered a new one (this time a different brand - Seagate). I will run `hd-idle` again tomorrow, if it doesn't survive it, I will immediately return it.

m33x, 2022-01-14:
This is what Windows 10 reports:
[image]

[image]

adelolmo, 2022-01-14:
Are you using a power adapter to run it? AFAIK Those external disk won't get enough power from the USB cable.

m33x, 2022-01-14:
I'm using the original USB-A to Micro B cable that was in the box with the HDD. Never had any issues with it. I would understand issues on the Pi, but not on my PC or Mac.

m33x, 2022-01-17:
I can report back, the new disk (Seagate Portable Drive, STGX5000400, UASP, Two year warranty, 100€) works as intended. `hd-idle` can spin down and spin up the disk. No issues what so ever. 

Given the fact that the old disk was my TimeMachine backup, I guess I have to thank you and `hd-idle` for spotting this faulty disk. Try to **avoid** Western Digital Elements 4TB (WDBHDW0040BBK) disks, they are not reliable (Sadly, the warranty expired in November 2021).

closed 2022-01-17 by m33x