adelolmo/hd-idle #69: Disk is in standby and hd-idle reports as spunDown as false

rajil, 2022-03-01:
Hello,

I am using the ata command to spindown the disk with an interval of 1800 (30 minutes). 

hdparm reports,

```
# hdparm -C  /dev/sdt

/dev/sdt:
 drive state is:  standby
```

However, hd-idle log says,

```
# journalctl -u hd-idle  |grep sdt |tail -1
Mar 01 07:23:29 nas hd-idle[7238]: disk=sdt command=ata spunDown=false reads=22594476 writes=64 idleTime=1800 idleDuration=120 spindown=2022-03-01T05:50:28 spinup=2022-03-01T05:51:28 lastIO=2022-03-01T07:21:29
```

The disk is formatted with LUKS.

adelolmo, 2022-03-01:
Thank you for reporting it and for giving details.

VadeInPace, 2022-03-20:
smartd 7.0 2019-05-21 r4917 [x86_64-linux-5.16.13-lp153.3.g9251332-default] (SUSE RPM)
Maybe the same/linked issue. `~> cat /etc/smartd.conf
...
DEFAULT -n standby,30,q
/dev/sda
/dev/sdb
/dev/sdb
/dev/sdd

~> sudo journalctl -b -u hd-idle.service -u systemd-suspend.service -u smartd.service
...
[sudo] password for root: 
-- Logs begin at Mon 2022-03-14 17:46:44 GMT, end at Sun 2022-03-20 11:54:04 GMT. --
Mar 20 10:53:15 localhost systemd[1]: Started hd-idle disk spindown service.
Mar 20 10:53:15 localhost hd-idle[1117]: hd-idle starting in nodaemon mode
Mar 20 10:53:15 localhost hd-idle[1117]:   disk: sdd timeout: 300
Mar 20 10:53:15 localhost hd-idle[1117]:   disk: sdc timeout: 300
Mar 20 10:53:15 localhost hd-idle[1117]:   disk: sdb timeout: 300
Mar 20 10:53:15 localhost hd-idle[1117]:   disk: sda timeout: 300
Mar 20 10:53:15 localhost hd-idle[1117]:   default timeout: 0
Mar 20 10:53:15 localhost systemd[1]: Starting Self Monitoring and Reporting Technology (SMART) Daemon...
Mar 20 10:53:15 localhost smartd[1142]: smartd 7.0 2019-05-21 r4917 [x86_64-linux-5.16.13-lp153.3.g9251332-default] (SUSE RPM)
Mar 20 10:53:15 localhost smartd[1142]: Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org
Mar 20 10:53:15 localhost smartd[1142]: Opened configuration file /etc/smartd.conf
Mar 20 10:53:15 localhost smartd[1142]: Drive: DEVICESCAN, implied '-a' Directive on line 32 of file /etc/smartd.conf
Mar 20 10:53:15 localhost smartd[1142]: Configuration file /etc/smartd.conf was parsed, found DEVICESCAN, scanning devices
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda, type changed from 'scsi' to 'sat'
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda [SAT], opened
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda [SAT], WDC WD40EFRX-68N32N0, S/N:WD-WCC7K1URKCCA, WWN:5-0014ee-264ba7edb, FW:82.00A82, 4.00 TB
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda [SAT], found in smartd database: Western Digital Red
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda [SAT], is SMART capable. Adding to "monitor" list.
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sda [SAT], state read from /var/lib/smartmontools/smartd.WDC_WD40EFRX_68N32N0-WD_WCC7K1URKCCA.ata.state
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb, type changed from 'scsi' to 'sat'
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], opened
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], WDC WD40EFRX-68N32N0, S/N:WD-WCC7K3FP1KRD, WWN:5-0014ee-2b936f586, FW:82.00A82, 4.00 TB
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], found in smartd database: Western Digital Red
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], is SMART capable. Adding to "monitor" list.
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], state read from /var/lib/smartmontools/smartd.WDC_WD40EFRX_68N32N0-WD_WCC7K3FP1KRD.ata.state
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc, type changed from 'scsi' to 'sat'
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], opened
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], WDC WD40EFRX-68WT0N0, S/N:WD-WCC4E1XXZ0TY, WWN:5-0014ee-20de6e65f, FW:82.00A82, 4.00 TB
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], found in smartd database: Western Digital Red
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], is SMART capable. Adding to "monitor" list.
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], state read from /var/lib/smartmontools/smartd.WDC_WD40EFRX_68WT0N0-WD_WCC4E1XXZ0TY.ata.state
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd, type changed from 'scsi' to 'sat'
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd [SAT], opened
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd [SAT], WDC WD40EFRX-68N32N0, S/N:WD-WCC7K1LDVZ6H, WWN:5-0014ee-2109636f0, FW:82.00A82, 4.00 TB
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd [SAT], found in smartd database: Western Digital Red
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd [SAT], is SMART capable. Adding to "monitor" list.
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdd [SAT], state read from /var/lib/smartmontools/smartd.WDC_WD40EFRX_68N32N0-WD_WCC7K1LDVZ6H.ata.state
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/nvme0, opened
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/nvme0, Samsung SSD 980 1TB, S/N:S649NX0RC29102M, FW:2B4QFXO7, 1.00 TB
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/nvme0, is SMART capable. Adding to "monitor" list.
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/nvme0, state read from /var/lib/smartmontools/smartd.Samsung_SSD_980_1TB-S649NX0RC29102M.nvme.state
Mar 20 10:53:15 localhost smartd[1142]: Monitoring 4 ATA/SATA, 0 SCSI/SAS and 1 NVMe devices
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdb [SAT], SMART Usage Attribute: 194 Temperature_Celsius changed from 128 to 127
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], SMART Prefailure Attribute: 3 Spin_Up_Time changed from 178 to 179
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], SMART Usage Attribute: 194 Temperature_Celsius changed from 129 to 127
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], previous self-test completed with error (read test element)
Mar 20 10:53:15 localhost smartd[1142]: Device: /dev/sdc [SAT], new Self-Test Log error at hour timestamp 14880
[... cut ...]
Mar 20 11:53:17 ASR5600G hd-idle[1117]: spinup: sdc
Mar 20 11:53:17 ASR5600G hd-idle[1117]: spinup: sdd

~> sudo hdparm -C /dev/sda /dev/sdb /dev/sdc /dev/sdd && date
[sudo] password for root: 

/dev/sda:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdc:
 drive state is:  standby

/dev/sdd:
 drive state is:  standby
Sun 20 Mar 11:54:14 GMT 2022
`
Look at the timestamp; Journatctl report disks spunup but hdparm report standby/spundown. hdparm is right, disks make NO sound, NO vibration at all.

adelolmo, 2022-07-24:
Hi @rajil 
Could you please have a look at this issue? https://github.com/adelolmo/hd-idle/issues/62
I'd like to understand if the cause of your problems is the same one or not.

adelolmo, 2022-07-30:
Fixed in [version 1.17](https://github.com/adelolmo/hd-idle/releases/tag/v1.17)

closed 2022-07-30 by adelolmo