adelolmo/hd-idle #70: smartctl --all /dev/sda make reads and no spin down

prchal, 2022-03-04:
Hi,
I use command periodically (60s) to get temperature of disk:
`smartctl --all /dev/sda | awk '/Temperature_Celsius/ {print $10}'`
but hd-idle reports reads and thus won't spin down:
`disk=sda command=scsi spunDown=false reads=39880915 writes=33187968 idleTime=77 idleDuration=8 spindown=0001-01-01T00:00:00 spinup=2022-03-04T16:28:55 lastIO=2022-03-04T16:32:00
disk=sda command=scsi spunDown=false reads=39880923 writes=33187968 idleTime=77 idleDuration=0 spindown=0001-01-01T00:00:00 spinup=2022-03-04T16:28:55 lastIO=2022-03-04T16:32:16
disk=sda command=scsi spunDown=false reads=39880923 writes=33187968 idleTime=77 idleDuration=8 spindown=0001-01-01T00:00:00 spinup=2022-03-04T16:28:55 lastIO=2022-03-04T16:32:16
disk=sda command=scsi spunDown=false reads=39880923 writes=33187968 idleTime=77 idleDuration=15 spindown=0001-01-01T00:00:00 spinup=2022-03-04T16:28:55 lastIO=2022-03-04T16:3`
If I spun disk down manually it stays down and reading temp works too.
Is there any solution of this?
Thanks Jiri

adelolmo, 2022-07-30:
Hi @prchal 
Would you mind to try out version [1.17](https://github.com/adelolmo/hd-idle/releases/tag/v1.17) and see if it makes any difference?

prchal, 2022-08-02:
Hi adelolmo,
I'm confused, looks like it didn't help:
`disk=sda command=scsi spunDown=true reads=109727262 writes=170619104 idleTime=360 idleDuration=3604 spindown=2022-08-02T09:26:03 spinup=2022-08-02T09:20:03 lastIO=2022-08-02T09:20:03`
run command:
`smartctl --all /dev/sda | awk '/Temperature_Celsius/ {print $10}'`
and then:
`sda spinup
disk=sda command=scsi spunDown=false reads=109727270 writes=170619104 idleTime=360 idleDuration=0 spindown=2022-08-02T09:26:03 spinup=2022-08-02T10:20:43 lastIO=2022-08-02T10:20:43`
but disk is not spinning.
BUT if I run hd-idle with long timeout and reading temp is stopped, it won't spin down:
`disk=sda command=scsi spunDown=false reads=109727256 writes=170619104 idleTime=3600 idleDuration=1741 spindown=0001-01-01T00:00:00 spinup=2022-08-02T07:33:57 lastIO=2022-08-02T07:59:59
disk=sda command=scsi spunDown=false reads=109727259 writes=170619104 idleTime=3600 idleDuration=0 spindown=0001-01-01T00:00:00 spinup=2022-08-02T07:33:57 lastIO=2022-08-02T08:30:00`
Something reads 3B every 1/2 hour?

adelolmo, 2022-08-22:
> Something reads 3B every 1/2 hour?

That's possible.
Could you please check https://github.com/adelolmo/hd-idle/issues/59#issuecomment-940886016?

prchal, 2022-08-29:
Not my situation, udisks2 never installed.

Daryes, 2022-09-09:
You need to change `--all` to implement `--nocheck=standby`

This should do the trick instead: `--info --health --attributes --tolerance=verypermissive -nocheck=standby --format=brief`
(taken from the telegraf monitoring agent => smart plugin)
Keep in mind smartctl will return an error (2) due to the drive sleeping.

prchal, 2022-09-13:
Did not help.

prchal, 2022-09-13:
Now I've got version 1.18 and it looks like any smartctl make reads.

pimlie, 2022-10-18:
Can confirm what @prchal mentioned above, even when the drive is already in standby (done manually by calling hdparm -y) it still counts reads. This means that you cannot use any smart monitoring (smartmontools or munin etc) with hd-idle at the moment:

Unfortunately Im not even sure how `hd-idle` could workaround this behaviour, seems to be either an upstream issue or just an unavoidable caveat. Maybe a setting like `ignore_read_threshold=N` to forcibly put a disk to idle if there are no writes and less then N reads within $idle_time?

```sh
~# cat /sys/block/sda/stat | awk '{ print $3 }'
32594596872
~# /usr/sbin/smartctl -n standby -i /dev/sda
smartctl 7.2 2020-12-30 (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(2)
~# cat /sys/block/sda/stat | awk '{ print $3 }'
32594596873
~# hdparm -C /dev/sda

/dev/sda:
 drive state is:  standby
```

chrishoage, 2022-12-10:
I'm not sure if this is the exact issue here but I wanted to share what fixed it for me. 

I was having this issue (smartctl causing reads on the device and not allowing spindown). I poked around the issues in the repo and came across this https://github.com/adelolmo/hd-idle/issues/38

I was not using partitions for my devices. I partitioned and formatted the partition of these devices and then hd-idle was able to detect partition reads only and it would splin down devices while smartctl would probe the device. 

Now this obviously won't be an option if you have a LUKS encrypted device, but something to consider for anyone reading this and having similar issues.

asquelt, 2023-03-29:
same here - partitionless drives are not spun down automatically, but when spinning them down (with -t) they stay in standby. something is incorrectly accounted as disk activity by hd-idle.

```
root@kkakdugi:~# lsb_release -a ; uname -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
Linux kkakdugi 5.15.60-1-pve #1 SMP PVE 5.15.60-1 (Mon, 19 Sep 2022 17:53:17 +0200) x86_64 GNU/Linux
```

other system with same configuration, but disks partitioned - hd-idle is correctly spinning them down.

```
root@kimchi:~# lsb_release -a ; uname -a 
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
Linux kimchi 5.15.60-1-pve #1 SMP PVE 5.15.60-1 (Mon, 19 Sep 2022 17:53:17 +0200) x86_64 GNU/Linux
```

adelolmo, 2023-11-02:
First of all I have to apologize for not being responsive at all:
I'm sorry!

After reading this issue (and some others) many times, I'm starting to grasp what's this about.
This is my current understanding:
* Monitoring disk tools (like smartmontools) increase reads in `/proc/diskstats`.
* If the disk has partitions the reads go to the disk level. hd-idle relays on the partitions' activity to determine activity. So all good here.
* If the disk has **no** partitions the reads go also to the disk level, but hd-idle relays on the disk to determine activity. Therefore hd-idle is mislead and believes there's disk activity.

prchal, 2023-11-02:
Hi, still not solved, but postponed to "sometimes".
You might be right, I don't have a partition on affected disk.
So, is there only solution to reformat disk with at least one partition?

adelolmo, 2023-11-02:
Currently I only see two options to proceed here:
1. Document that disk without partitions are not supported (known shortcoming)
2. Introduce a threshold to ignore reads

This is all mentioned here: https://github.com/adelolmo/hd-idle/issues/70#issuecomment-1281952416

Where _option 1_ would be the easiest to implement from my side, I see the value for users to implement _option 2_.
The _option 2_ requires introducing the concept of threshold. 
If hd-idle puts defining the threshold on the user side, I see how questions/issues will pile here (I don't like giving support to niche scenarios).
On the other hand, if hd-idle determines the threshold... it better is a damn good algorithm, because it might have to be calculated dynamically for every case :man_shrugging: 

> So, is there only solution to reformat disk with at least one partition?

@prchal Currently yes, that's the only solution if you want to keep disk monitoring tools.