adelolmo/hd-idle #74: Build from source: go get gives error message "depreciated use go install instead"

T-6, 2022-07-23:
When trying to build from source on openSUSE I get the error message, that "go get" is depreciated since go version 1.17 and "go install" shall be used instead. When using go install, I need to provide credentials for github (which I have) and for the repository (which I do not have).
Would be nice if the documentation could be updated on how to build from source with a go version of 1.17 and above.

adelolmo, 2022-07-23:
Can you please explain how you're building it?

T-6, 2022-07-23:
When I run `go get github.com/adelolmo/hd-idle` I get the mentioned error message.

adelolmo, 2022-07-23:
If you only what to build the program, but not the debian package, you could follow this steps:
```
git clone https://github.com/adelolmo/hd-idle.git
cd hd-idle
go build
```
You should get the binary file `hd-idle`.

Let me know if you have any problem with the above steps.

T-6, 2022-07-23:
That worked perfectly. Thanks a lot!!!
And thanks a lot for the port of hd-idle.

closed 2022-07-23 by adelolmo