adelolmo/hd-idle #77: 'checking skew' is spammed in syslog

auanasgheps, 2022-08-09:
Hi,
hd-idle is prints in the syslog the following messages, continuously since some days.
What is going on? :(
```
Aug  9 13:36:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sde now: 2022-08-09 13:36:41 lastNow: 2022-08-09 13:35:41, now-lastNow: 1m0.051608267s
Aug  9 13:36:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sda now: 2022-08-09 13:36:41 lastNow: 2022-08-09 13:35:41, now-lastNow: 1m0.051607991s
Aug  9 13:36:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdb now: 2022-08-09 13:36:41 lastNow: 2022-08-09 13:35:41, now-lastNow: 1m0.051610526s
Aug  9 13:36:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdd now: 2022-08-09 13:36:41 lastNow: 2022-08-09 13:35:41, now-lastNow: 1m0.051609478s
Aug  9 13:36:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdc now: 2022-08-09 13:36:41 lastNow: 2022-08-09 13:35:41, now-lastNow: 1m0.051608892s
Aug  9 13:37:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sde now: 2022-08-09 13:37:41 lastNow: 2022-08-09 13:36:41, now-lastNow: 1m0.059241686s
Aug  9 13:37:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sda now: 2022-08-09 13:37:41 lastNow: 2022-08-09 13:36:41, now-lastNow: 1m0.059241686s
Aug  9 13:37:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdb now: 2022-08-09 13:37:41 lastNow: 2022-08-09 13:36:41, now-lastNow: 1m0.059241686s
Aug  9 13:37:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdd now: 2022-08-09 13:37:41 lastNow: 2022-08-09 13:36:41, now-lastNow: 1m0.059241686s
Aug  9 13:37:41 nas hd-idle[1125]: checking skew. skewtime: 3m0s disk: sdc now: 2022-08-09 13:37:41 lastNow: 2022-08-09 13:36:41, now-lastNow: 1m0.059241686s
```

adelolmo, 2022-08-09:
I think this is a leftover from the release candidate you tested. 
Could you please remove it and install it again from the repository?

auanasgheps, 2022-08-10:
Removed and reinstalled, still spamming

adelolmo, 2022-08-10:
OK. I'll patch it and release it next week.

auanasgheps, 2022-08-11:
All right, let me know if you need testing.

adelolmo, 2022-08-17:
Hi @auanasgheps 
The newest release https://github.com/adelolmo/hd-idle/releases/tag/v1.18 fixes the issue.

I made the v1.17 release from a branch instead of doing it from master. This is why the annoying log made its way into the package.

auanasgheps, 2022-08-17:
Thank you for the update!
Unfortunately my server will be unavailable until early September, so let's close this issue and I will reopen if needed.

One last question. I see this entry
`Complete the list of targets for the service restart`

Is this another attempt to fix resume after standby/hibernation?

adelolmo, 2022-08-17:
> Is this another attempt to fix resume after standby/hibernation?

I'm not sure what would help in this case, but for completeness [I added to the service all suspend/hibernate related targets](https://github.com/adelolmo/hd-idle/blob/master/debian/hd-idle.service#L4).

auanasgheps, 2022-09-11:
We can close this issue, the 1.17 version has resolved the issue. Thanks!

closed 2022-09-11 by auanasgheps