adelolmo/hd-idle #79: tweaks in default file for building the drive list

Daryes, 2022-09-09:
Hello,
Didn't know about this rewrite in go of hd-idle, nice of you.

This post is not really an issue, but more some ideas about the building of the drive list.
Since I started to use the original hd-idle, I've rewritten the default file to autodetect the existing drives at boot.
This change is working without trouble since many years, even with the presence of LUKS volumes or ZFS (if not both).

Here's the content: ( /etc/default/hd-idle )
```
# start hd-idle automatically : true | false
START_HD_IDLE=false

# hd-idle command line options
# Options are:
# (...)

# drive timeout
HDD_Timeout=3600

# build the hdd list
for sCurrentDrive in $( ls -1 /dev/disk/by-id/ | grep -i '^ata-' |egrep -v -i '\-part|SSD|_(BD|DVD|CD)' ); do
        HD_IDLE_OPTS="${HD_IDLE_OPTS} -a /dev/disk/by-id/$sCurrentDrive -i ${HDD_Timeout}"
done

# activate the execution
[ ! -z "$HD_IDLE_OPTS" ] && START_HD_IDLE=true
```
Obviously the systemd unit can't handle this, but the sysinit script does.

Dunno what are your plans about the future of hd-idle, but it would be interesting to have an automatic drive list integrated into a new argument taking over if no drive are provided, with maybe the possibility to filter out some drives id / names.

Feel free to close this issue.

adelolmo, 2022-09-11:
Hi @Daryes 

So far hd-idle is design to have a default idle time and an idle time specific per given disk (which overrides the default).

```
HD_IDLE_OPTS="-i 600 -c ata -a /dev/disk/by-partlabel/disk2 -c scsi -a /dev/disk/by-label/parity1 -c scsi -l /var/log/hd-idle.log"
```
In the example above, the defaults are `-i 600` and `-c ata`.

From my understanding of your comment, this is not enough for your usecase? 
I'd like to understand it better.

Daryes, 2022-09-11:
Nothing to do with the default values, that was more a help for users that interested by hd-idle but not used to navigate with the subtleties under /dev/disk. The grep filter I've used currently handle most of the situation to collect all connected drives only while excluding ssd, nvme and cd drives.  Granted, it's only for drives detected as sata type by the system.
As said, it's more to give ideas than a real issue.

For the timeout duration of 1h, it's related to the number of disks on my NAS : the usage is usually only in a range of hours. As there are multiples drives, I didn't want them getting spinned up and down constantly.
Especially as they are set to use a staggered order to prevent excessive load on the power supply (drives use more power on start up), and the wake up sequence takes some seconds due to the number of drives.

So the large timeout is to ensure they go to idle mode only when nobody will access them for enough time (night, or work day for example)

closed 2023-05-07 by adelolmo