adelolmo/hd-idle #80: Can't spin down anymore after a spin up

Tuweky, 2022-09-13:
Hi,

I have a Seagate STKM5000400 attached to my Raspberry Pi 4 running Raspberry PI OS Lite 64-bit with hd-idle version 1.18 installed. Currently I have the hd-idle service disabled and do some manual spin down tests. Yet I am only able to let the disk spin down once. After letting the disk spin up again (by mounting/accessing it), it won't spin down anymore at all until I reboot the system or re-plug the usb device.

When I apply the the commands as listed below, the second time that I try to call hd-idle, the disk won't spin down anymore.
```
hd-idle -t /dev/disk/by-id/usb-Seagate_Expansion_SW_00000000NAC76CES-0\:0 // successful spin down
mount /dev/disk/by-id/usb-Seagate_Expansion_SW_00000000NAC76CES-0\:0-part1 /mnt // disk spins up again
hd-idle -t /dev/disk/by-id/usb-Seagate_Expansion_SW_00000000NAC76CES-0\:0 // no spin down
```
Applying the `-d` argument to `hd-idle` doesn't output any log on the terminal. Also at `/var/log/messages` there isn't anything listed about the disk or `hd-idle` actions. The second `hd-idle` call does return `0` when applying `echo $?`.

Unmounting the disk either doesn't allow me to let it spin down again, only a power cycle of the system or re-plugging the usb disk makes it possible to spin-down it down again.

Any idea's or suggestions how I can tackle this issue or further analyse it?

rabelux, 2022-09-13:
Which version are you running?

Tuweky, 2022-09-13:
> Which version are you running?

version 1.18 as mentioned

raspby4, 2022-12-30:
Dear all,

I can confirm the described malfuntion and hopefully I can contribute some hints to solve this issue.

In short: Hard disk spins down as expected the first time. When accessing the hard disk again it spins up and read / write actions are possible (though the indicator LED is not lit again). From this time on the hard disk will never spin down again until system is rebooted.

Results of further testing:

1) There is a tool called Seagate Dashboard which enables the user to set a spin down timer. Connecting the hard disk to a linux based router it works as intended. Connected to the raspberry this does not work.
**Conclusion**: General capabilities work as expected but some system specific behaviour seems to cause the malfunction.

2) On the raspberry I tried to send SCSI commands using SDPARM.
sudo sdparm --readonly --command=stop /dev/sda --> spin down
sudo sdparm --command=start /dev/sda --> spin up
These commands work fine in general. However, if I spin down and access the hard disk for read / write actions they are performed correctly but like with hd-idle the LED is not lit again and the hard disk won't spin down if I send the spin down command a second time. Once I send the spin up command I can spin down the disk with the spin down command properly.
**Conclusion**: Accessing data on the hard disk does not wake up the disk properly. A disk not waken properly cannot be spun down.

3) Same behaviour as with the spin down command can be achieved with "hd-idle -t /dev/sda". If disk is waken up with spin up command afterwards, the disk can be spun down by sending this hd-idle command a second time. If disk is not waken up then a spin down after accessing data is not possible.

System environment: Raspberry 4 with Raspbian Buster 32-bit desktop version and kernel version "5.10.103-v7l+". HDD is a Seagate Expansion 4TB connected on USB3 port.

adelolmo, 2023-01-25:
@raspby4  My environment is similar to yours, but running on a server with mostly Western Digital disks.

I'm pretty lost here and won't be able to give any useful feedback.
I can only ask both of you @Tuweky @raspby4  to try out using the ATA command to spin down the disks.

raspby4, 2023-02-12:
Dear all,

Finally after spending hours for research (as usual for Linux ;) I think I found the root cause: It's an incompatibility with the UAS protocol. A workaround is described here ([https://www.smartmontools.org/wiki/SAT-with-UAS-Linux]). After applying that I was able to spin down my hard drive using hdparm. Not yet confirmed with hd-idle, but probably it's the same.

Have a nice day!

adelolmo, 2023-04-20:
BTW This PR https://github.com/adelolmo/hd-idle/pull/85 might solve this issue, could any of you test hd-idle [v1.20](https://github.com/adelolmo/hd-idle/releases/tag/v1.20)?

closed 2023-04-20 by adelolmo