adelolmo/hd-idle #81: No spindown after resume in v1.18.

WonderSkydive, 2022-10-04:
Hello. First of all, congratulations for the continuity of this project and for having released version 1.18.

I have installed this version on a fresh install of Debian 11.5, I couldn't build the package myself using dpkg-buildpackage (several errors, despite installing golang-go, dkpg...always error on make. So I used the already created file in .deb
Congratulations! This version brings what is necessary so that, once we resume the computer from sleep, hibernation or hybrid sleep, the hd-idle service continues to work!
But... the error that I have detected:
We start pc for the first time, without doing anything, the hard drives go to spindown without problems, we suspend pc, when resuming, the hd-idle service continues to run but ... it does not spin down the hard drives, so that it spins them down , you must access the hard disk, so if, after the configured time, it takes it to spindown, but in my case, I have 3 hard disks, it only takes spindown to the one I access, for example, if I access 1, 2 and 3 stay up, if then I access 2, 3 stays up. This always, after resuming from sleep, if I want it to take the hard drives back to spindown, I must access all 3 hard drives, if I only access 1 or 2, the remaining one or more, does not take them to spindown.

If I install/run hd-idle-restart-resume.service which comes from this patch: https://sourceforge.net/p/hd-idle/patches/2/ (on version 1.18 here) it works every time.

But I don't like to mix a new version like this improved with an old patch.

I hope I was able to explain myself.

Thank you for this work and greetings.

adelolmo, 2022-10-11:
Duplicate of https://github.com/adelolmo/hd-idle/issues/66

Would you like to discuss here your issues building the package? Otherwise I'll close it as is a duplicated of a known issue.

WonderSkydive, 2022-10-11:
In a certain way the problem is different, I am aware that I opened the other issue, but what happened to me in the other issue is different from what happens to me in this one.
In the other, once I suspended, the hard drives never went into spindown on resume.
In this new version yes, but only if I access the hard drives, if not, they don't go to spindown.
I hope I have been able to explain well the difference between the other version and this one.
If I'm wrong and the same thing happened in the other version, apologies.
I have had the feeling that this improvement exists in this version, but as I mentioned, if I am wrong, apologies.

About building the package, I have no problem because I use the deb already compiled by you, but we can look at solving this too, everything that is improving it or solving problems, perfect.

adelolmo, 2023-01-19:
@WonderSkydive 
First of all: sorry for the late replay. I find very hard to find motivation to work on this project these days :/

I've done a bit of testing with v1.18 and this is what I got:
1. all disks are spinning
2. suspend machine
3. after few minutes (seconds in my testing scenario) I see this in the log file:
```
date: 2023-01-19, time: 16:53:59, disk: sda, assuming disk spun up after long sleep
date: 2023-01-19, time: 16:53:59, disk: sdb, assuming disk spun up after long sleep
date: 2023-01-19, time: 16:53:59, disk: sdc, assuming disk spun up after long sleep
date: 2023-01-19, time: 16:53:59, disk: sdd, assuming disk spun up after long sleep
```
and this in the program output (syslog):
```
disk=sda command=scsi spunDown=false reads=271758984 writes=224511336 idleTime=5 idleDuration=5 spindown=2023-01-19T16:52:19 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sdb command=scsi spunDown=false reads=4095688 writes=2073192 idleTime=5 idleDuration=5 spindown=2023-01-19T16:52:19 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sdc command=scsi spunDown=false reads=765554100 writes=81564633 idleTime=5 idleDuration=5 spindown=2023-01-19T16:52:19 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sdd command=scsi spunDown=false reads=12742 writes=3 idleTime=5 idleDuration=5 spindown=2023-01-19T16:52:19 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
sda spindown
disk=sda command=scsi spunDown=true reads=271758984 writes=224511336 idleTime=5 idleDuration=5 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
sdb spindown
disk=sdb command=scsi spunDown=true reads=4095688 writes=2073192 idleTime=5 idleDuration=5 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
sdc spindown
disk=sdc command=scsi spunDown=true reads=765554100 writes=81564633 idleTime=5 idleDuration=5 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
sdd spindown
disk=sdd command=scsi spunDown=true reads=12742 writes=3 idleTime=5 idleDuration=5 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sda command=scsi spunDown=true reads=271758984 writes=224511336 idleTime=5 idleDuration=6 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sdb command=scsi spunDown=true reads=4095688 writes=2073192 idleTime=5 idleDuration=6 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
disk=sdc command=scsi spunDown=true reads=765554100 writes=81564633 idleTime=5 idleDuration=6 spindown=2023-01-19T16:54:04 spinup=2023-01-19T16:53:59 lastIO=2023-01-19T16:53:59
```

This tells me that it works as intended.

Could you please check if you get the message `assuming disk spun up after long sleep` in the log file?

msz-kp, 2023-01-27:
i have the same problem
..what i noticed
now.Sub(lastNow) in updateState(tmp DiskStats, config *Config) returns incorrect value
now.Sub(lastNow) > config.SkewTime is always false

quick digup tells me that .Sub uses monotonic clock which on some systems may stops when the computer is in S3 sleep (https://github.com/golang/go/issues/36141)
if i change updateState to:
`
	duration := time.Now().Unix() - lastNow.Unix()
	if duration > int64(config.SkewTime/1000000000) {
`
everything works as excepted

compiled with go 1.19.4

adelolmo, 2023-01-29:
This is a great finding.
I'll make sure it's implemented in the next release.

closed 2023-02-06 by adelolmo

adelolmo, 2023-02-09:
Hi @WonderSkydive and @msz-kp 
Could you please try out the new version [v1.19](https://github.com/adelolmo/hd-idle/releases/tag/v1.19) and see if it solves this issue?

WonderSkydive, 2023-02-09:
Hi @adelolmo 
My apologies for the delay in responding to the other reply. I saw your publication and I was away for a few days (due to force majeure without being able to attend to anything), a few days ago I reincorporated and I have been catching up on everything pending, I have also read the updates that you have posted about it.
Without extending myself further, I did not answer at the beginning due to force majeure.
For version 1.18, in my case I solved it with the mentioned patch.
Today is Thursday night here, tomorrow Friday I have a long day at work. At the latest, this weekend when I have nothing to do, I completely uninstall 1.18, install 1.19 and report results.
Thanks and regards.

WonderSkydive, 2023-04-04:
Excuse me for the delay in answering, I wanted to try it well and in recent days, I lacked time for everything.
Say that ... simply great!
I made a complete uninstall of 1.18 and everything related to HD-IDLE 1.18 for its operation. I left everything clean of trace.
I installed this new version 1.19, I configured and ... simply great!
Now, it works 100%.
At any time, the discs pass to Spin Down with the scheduled time, after awakening the sleeping, or hibernation PC, or in my case, now I use Sleep Than Hibernate.
At all times, hard drives go to Spin Down after the time configured.
Now he does not have the failure that, after resuming PC, he had to access the hard drive so that later, he passed Spin Down.
Now it simply works great.
I have been trying about 15 to be able to give adequate feedback, every day I observe it, the comment, perfect!
Thanks for your work @adelolmo

PR #91 (2023-02-06): Use UNIX time to calculate skew interval
Fixes #81
commit d53a8a3f (2023-02-06): fixes #81 Use UNIX time to calculate skew interval