adelolmo/hd-idle #86: Question about hd-idle functional logic?  Device discovery is done when?

Jibun-no-Kage, 2022-12-05:
Question about hd-idle functional logic?  Device discovery is done when?  Scanning the fstab is one thing, but as I understand it, hd-idle only scans the fstab file on service start/restart, or am I mistaken?

Why my question,  have various usb storage devices, which are disk based, that are rotated based on use.  When in use, they maybe connected to the given system for several hours, often days, or longer.  But these devices are not explicitly defined in the fstab, but mount or unmounted via UDEV rules that execute a script that create the mount on add event, and unmount done on remove event for the given device.

May I correct or mistaken that hd-idle will not automatically see they UDEV/script controlled/supported devices?

adelolmo, 2023-01-04:
Hi @Jibun-no-Kage 
Device discovery is done by [reading the file /proc/diskstats](https://github.com/adelolmo/hd-idle/blob/master/diskstats/snapshot.go#L104) and fstab is not involved in the process.
hd-idle is constantly monitoring /proc/diskstats and it will pick any device additions automatically.

Jibun-no-Kage, 2023-01-04:
Excellent, thank you.

closed 2023-01-04 by Jibun-no-Kage