adelolmo/hd-idle #87: Unconditional spindown command

gusr, 2023-01-06:
I have a system with some Seagate drives and some Western Digital drives. Hd-idle works as expected with all drives, and they spin down properly, entering the "standby" state as indicated by hdparm.

But after many hours of no activity, the Seagate drives spin up again on their own accord, now shown as "drive state: unknown" by hdparm. At this point hd-idle still believes that the drives are spun down, and they will therefore remain spun up forever, unless there is some activity on the drives, at which point hd-idle will again spin them down properly, after the set amount of time.

I am not sure what the cause of this behavior is; I have upgraded the drive firmware with no change. But hd-idle fails to spin down the drives because it assumes that they will remain spun down if they have been asked to, if there was no subsequent activity as seen by the OS.

I suggest adding a per-drive configuration option to unconditionally call spindownDisk(), even if ds.SpunDown == true. Alternatively, one could check for the "unknown" power state before that point and set ds.SpunDown to false in that case.

I could write some code for this, but I am not sure which of the logging and bookkeeping code surrounding the call to spindownDisk() should be run in this scenario, and also if there are some other problems with this brute force approach?

adelolmo, 2023-01-06:
Hi @gusr 
Could you please share here your `/proc/diskstats` while there's no activity (and disk are spun down) and once the disk spin up with "drive state: unknown" ?
My guess is that there's some activity register on the disk level but not reflected on any partition.

This is a common issue for some users: https://github.com/adelolmo/hd-idle/issues/80 https://github.com/adelolmo/hd-idle/issues/70

gusr, 2023-01-07:
I logged the output of "grep sdb /proc/diskstats" and "hdparm -C /dev/sdb | tail -n 1" every hour overnight. 

Here is the output for 16 hours, only showing the diskstats when they change:

   8      16 sdb 20456593 26574 3655261112 57578829 12643303 19272 1069856872 8497148 0 56747532 87385047 0 0 0 0 278951 21309070
   8      17 sdb1 20456168 26574 3655235768 57577485 12643303 19272 1069856872 8497148 0 56746496 66074633 0 0 0 0 0 0
   8      18 sdb2 224 0 16640 759 0 0 0 0 0 768 759 0 0 0 0 0 0

/dev/sdb:
 drive state is:  unknown

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby
 
   8      16 sdb 20456667 26574 3655268896 57587051 12643303 19272 1069856872 8497148 0 56755776 87393269 0 0 0 0 278951 21309070
   8      17 sdb1 20456184 26574 3655239352 57577518 12643303 19272 1069856872 8497148 0 56746548 66074666 0 0 0 0 0 0
   8      18 sdb2 280 0 20800 8928 0 0 0 0 0 8940 8928 0 0 0 0 0 0

/dev/sdb:
 drive state is:  unknown

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  standby

/dev/sdb:
 drive state is:  unknown

/dev/sdb:
 drive state is:  unknown

/dev/sdb:
 drive state is:  unknown

/dev/sdb:
 drive state is:  unknown

So the drive starts out in the "unknown" state, is put to sleep, has some activity, goes to unknown, and is put to sleep again. After a few hours, it then audibly spins up without any change in /proc/diskstats, along with entering the "unknown" state again.

Meanwhile, in the hd-idle debug log, the disk activity associated with the change in /proc/diskstats above was detected, and hd-idle properly spun down the drive afterwards. But hd-idle is completely oblivious to the second transition to "unknown", and the last debug output for the disk reads:

jan 07 14:19:04 box hd-idle[1198905]: disk=sdb command=scsi spunDown=true reads=3655260152 writes=1069856872 idleTime=1800 idleDuration=28632 spindown=2023-01-07T06:51:52 spinup=2023-01-07T06:20:51 lastIO=2023-01-07T06:21:51

gusr, 2023-01-22:
To sum up, something is spinning up these drives after many hours of inactivity:

- It is not related to file system activity; these drives are part of a ZFS pool, together with others that don't spin up.
- It is not related to smartmontools, which is properly ignoring the drives while they are in standby, plus I tried turning that service off altogether.
- No trace is left in /proc/diskstats.
- I have updated the drives to the latest firmware.

Any ideas what might be happening, and if so how hd-idle could detect it?

adelolmo, 2023-01-25:
hd-idle won't be able to help you debugging your issue.
I never had this situation happening to me. My disks run connected to a Raspberry Pi running Raspberry Pi OS and it always behaves nicely.

closed 2023-04-14 by adelolmo