adelolmo/hd-idle #88: Meaning of "Use disk partitions or device mapper to calculate activity" ?

malvinas2, 2023-01-08:
Hello, 
what does the extra feature "Use disk partitions or device mapper to calculate activity" exactly mean?

I am using `Ubuntu 22.04.1 LTS`, `kernel 5.15.0-56` and installed the **original** version 1.5 of `hd-idle`, written by Christian Müller. 
Everything seems to work, all six hard drives on my small home server go into sleep mode after a set time. 
What happens if I combine all six disks into a single virtual disk using LVM? The mentioned extra feature deals with kernels newer than 5.4 LTS, which is the case in Ubuntu 22.04. So would I get in trouble with the old version of Christian? 

Why should I use `/dev/mapper/luks-0123456890` or `/dev/disk/by-id/abcdefgh` instead of just e.g. `sda` ? 
If I were to use `/dev/mapper`, the whole cluster (=all drives) would either be running or spinned-down. In my opinion this makes no sense, for example a file in use could be on drives 4 and 5, so that drives 1, 2, 3 and 6 could therefore go into standby mode. 
(that's how it works in my other home server, which is still running `Ubuntu 16.04 LTS`). 

A parallel installation of both hd-idles (original+remake) for test purposes is probably not possible, since the files used have the same names, right?

adelolmo, 2023-01-25:
Hope the answers help.

> what does the extra feature "Use disk partitions or device mapper to calculate activity" exactly mean?

It means that it can spin down disks encrypted via LUKS.

> What happens if I combine all six disks into a single virtual disk using LVM? 

Only disk with no read/write activity will spin down. I not 100% sure of how LVM works, so take this answer with care.

> So would I get in trouble with the old version of Christian?

I don't see why you would.

> Why should I use /dev/mapper/luks-0123456890 or /dev/disk/by-id/abcdefgh instead of just e.g. sda ?

Because the disk id won't change and the `sda` assignation might.

> A parallel installation of both hd-idles (original+remake) for test purposes is probably not possible, since the files used have the same names, right?

Yes, you are right.

malvinas2, 2023-01-30:
Based on your answer I didn't understand what actually changed now or what the benefit of the new feature is. 
In particular, when you write "I don't see why you would (get in trouble with the old version)", that questions the point of all the changes ;-) 
After reading the related pull requests, I would summarize the issue as follows: For some unknown reason, the kernel's behavior has changed. In newer kernels disk monitoring tools -e.g. even temperature query of hddfancontrol- change read/write values on disk level. To avoid this, disk activity was calculated by watching read/write changes on partitions (sda1 instead of sda). 
Later on someone explained that it is not sufficient to just watch normal partitions, but also encrypted ones (dm-x partitions). 

1. Should the documentation be changed to use partitions everywhere? It would be consistent that way.
Instead of 
`hd-idle -i 0 -a sda -i 300 -a sdb -i 1200`
the example would be something like
`hd-idle -i 0 -a sda1 -i 300 -a sda2 -i 300 -a sdb1 -i 1200
`

2. As you already wrote yourself, there will be "many issues opened asking for further explanations".  :-) 
Maybe @spasche and @V33m can name more application examples.

For example, my old media server still running fine with Ubuntu 16.04 and hdidle 1.5 is configured as follows: 
```
disc_1 \
        > raid_md1 - crypt-raid_md1 - pv    \
disc_2 /                                                      | 
                                                                  |
disc_3 \                                                      \
        > raid_md2 - crypt-raid_md2 - pv      >   LVM
disc_4 /                                                      /
                                                                  |
disc_5 \                                                      |
        > raid_md3 - crypt-raid_md3 - pv    /
disc_6 /
```
According to the remarks so far ("The whole external drive is encrypted") an order like `RAID --> LVM --> LUKS --> ext4` would be a problem (?). 
Here the order is `RAID --> LUKS --> LVM --> ext4`. 

3. Although it's not your problem: Is the changed behaviour of the kernel a bug or a feature? If it is some kind of (changed) feature, I would have expected a notification from the developers. 
What about the [comment](https://github.com/desbma/hddfancontrol/issues/28#issuecomment-814410789) "passing partitions can be confusing because it can lead to spinning down a drive that is currently active on another partition."?

closed 2023-05-07 by adelolmo