adelolmo/hd-idle #93: Init scripts blocked

Zweikeks, 2023-02-10:
Installing hd-idle_1.18_armhf.deb blocks further init scripts after hd-idle is started during the boot process in runlevel 2.
Since hd-idle does not return it must be started via start-stop-daemon with --background added.
See /etc/init.d/hd-idle attached.
(Especially important because getty is no longer started. Thus logging in via serial is not possible after installing hd-idle.)

[hd-idle.zip](https://github.com/adelolmo/hd-idle/files/10710557/hd-idle.zip)

adelolmo, 2023-02-10:
Hi @Zweikeks 
This is something doable.
Duplicate of #53

Zweikeks, 2023-02-10:
Thank you for taking care. Thank you for maintaining hd-idle.

adelolmo, 2023-02-16:
Hi @Zweikeks 
Could you please try out [v1.20](https://github.com/adelolmo/hd-idle/releases/tag/v1.20)?

Zweikeks, 2023-02-17:
Hello Andoni,

I have tried the package hd-idle_1.20_armhf.deb. Works like a charm!
It does _not_ hang during installation with dpkg -i (like 1.18 did).
It does _not_ block the init scripts during the boot process.
Hard disk goes into standby after 20min with my configuration.
All good.

Minor thing:
In version 1.20 the manpage hd-idle.8.gz is missing.
(Can also be checked with dpkg-deb -c hd-idle_1.20_armhf.deb without installing.)

Thanks a lot for making the change!

adelolmo, 2023-02-17:
Thank you for your feedback @Zweikeks 

I've created a new v1.20 fixing also the missing man page.

Zweikeks, 2023-02-17:
Just great!

closed 2023-02-17 by adelolmo