adelolmo/hd-idle #95: Can't install on Raspberry Pi

remirr, 2023-03-26:
Hi there,

I am not bad at installing on Raspberry Pi but I am not getting anywhere with hd-idle. I have a Seagate USB drive that never spins down and I need help. 

The drive is used for a remote backup everyday at 4pm, after the backup it should spin-down then spin-up again the next day.

I have used git clone https://github.com/adelolmo/hd-idle to download hd-idle. It seems to have downloaded hd-idle to a folder under my home drive but from then on I have no idea what to do....

Any help would be appreciated. 

Thank you

remirr, 2023-03-26:
Tried using the Debian install instructions and get

@raspberrypi:~/hd-idle $ dpkg-buildpackage -a armhf -us -uc -b
dpkg-buildpackage: info: source package hd-idle
dpkg-buildpackage: info: source version 1.20
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Andoni del Olmo <andoni.delolmo@gmail.com>
dpkg-buildpackage: info: host architecture armhf
 dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9) golang-go:native (>= 1.3.3) dh-golang
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)

remirr, 2023-03-26:
Ok managed to install it with the package. When running I get this and it looks like the command never completes....

remigi@raspberrypi:~ $ sudo apt install ./hd-idle_1.20_armhf.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'hd-idle' instead of './hd-idle_1.20_armhf.deb'
The following packages were automatically installed and are no longer required:
  easy-rsa libfuse2 libpkcs11-helper1 opensc opensc-pkcs11
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  hd-idle
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/1,235 kB of archives.
After this operation, 2,558 kB of additional disk space will be used.
Get:1 /home/remigi/hd-idle_1.20_armhf.deb hd-idle armhf 1.20 [1,235 kB]
Selecting previously unselected package hd-idle.
(Reading database ... 109432 files and directories currently installed.)
Preparing to unpack .../remigi/hd-idle_1.20_armhf.deb ...
Unpacking hd-idle (1.20) ...
Setting up hd-idle (1.20) ...
Processing triggers for man-db (2.9.4-2) ...
remigi@raspberrypi:~ $ hd-idle
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, defaultPowerCondition=0, debug=false, logFile=, devices=

remirr, 2023-03-26:
So after 15 mins I got this: 

remigi@raspberrypi:~ $ hd-idle
symlinkPolicy=0, defaultIdle=600, defaultCommand=scsi, defaultPowerCondition=0, debug=false, logFile=, devices=
sda spindown
cannot spindown scsi disk /dev/sda:
open /dev/sda: permission denied

adelolmo, 2023-03-27:
Let me know if you have any problem running hd-idle after reading [the documentation](https://github.com/adelolmo/hd-idle#run-hd-idle) on the topic.

closed 2023-03-29 by adelolmo

remirr, 2023-03-30:
Hello @adelolmo , I tried following these instructions, installation ended up going fine and it seems to be running but the hard drive has never actually switched off. How do I go about looking at the logs to see what is actually happening? Thank you

adelolmo, 2023-03-30:
> How do I go about looking at the logs to see what is actually happening?

That would be here: [Understand the logs](https://github.com/adelolmo/hd-idle#understand-the-logs)

remirr, 2023-03-30:
Ok, so now I am getting: 
sda spinup
sda spindown
cannot spindown scsi disk /dev/sda:
open /dev/sda: permission denied

adelolmo, 2023-03-30:
Hi @remirr 
I don't want to sound rude, but it would be way easier if you read the documentation to begin with. My time is limited
> Ok, so now I am getting:
sda spinup
sda spindown
cannot spindown scsi disk /dev/sda:
open /dev/sda: permission denied

remirr, 2023-03-30:
Ok so I started from scratch and was able to get it working, the drive spins down but then the power pack makes a high pitched noise that sounds like an ambulance. It does this until you unplug the drive and plug it back in to spin it up. 

Also I am using the drive for remote backup of a Synology Nas at the remote location. The backup now says offline when the drive spins down so it seems that this won’t work as it sees the drive as disconnected from the RPI. It’s a shame because having the drive spin 24/7 for a once a day backup that lasts 30 mins seems excessive! 

Thanks for the awesome package but seems I am out of options