adelolmo/hd-idle #97: Possible to spin-down based on Schedule?

remirr, 2023-05-04:
Hi there,

I am using a Rpi 400 with an 18TB Seagate Extension drive that does not spin down automatically. I am using this for a offsite remote backup of a Synology NAS which has a tunnel over wireguard VPN. The problem is when the HDD spins down the NAS see's it as 'disconnected' and it disappears from the RaspberryPi MNT folder (literally disconnects from the Rpi). So is it possible to set a schedule for the RaspberryPi to spin up the HDD 30 mins before the scheduled backup and then spin it back down at say 9pm? then do that everyday? 

Thank you!

remirr, 2023-05-06:
so would i write something then delete it again as part of the operation? Do you have a bit of code I could use?

adelolmo, 2023-05-06:
As I already said: Any read/write operation should do the trick and spinup the disk.

You could try googling a little bit. I don't see how this question is related to hd-idle anymore.

closed 2023-05-07 by remirr