adelolmo/hd-idle #98: Logfile won't be created

opc241, 2023-05-06:
/etc/default/hd-idle contains the following line (regarding logfile): 
`HD_IDLE_OPTS="-i 0 -a /dev/disk/by-id/usb-TO_Exter_nal_USB_3.0_2015033100081-0:0 -i 1200 -l /home/homecontrol/hd-idle.log"`

The problem is, after spindown/spinup no logfile is created. Any idea why? I tried different paths and the actual path configured is not a path on the hdd which i want to spin down.

adelolmo, 2023-05-07:
Could you please have a look at the [log file documentation](https://github.com/adelolmo/hd-idle#log-file)?
Let me know if the problem persists afterwards.

opc241, 2023-05-08:
Before i posted this, i already had a look on this documentation, but i can't see the reason what i am doing wrong. The disk spins up/down but no logfile is created.

adelolmo, 2023-05-08:
You could try creating an empty log file before running hd-idle.
e.g. `touch /home/homecontrol/hd-idle.log`

Also, the daemon is running the service as `root`. Make sure that the `root` user has permissions to create/write into the file.

opc241, 2023-05-08:
> Make sure that the `root` user has permissions to create/write into the file.

Thanks, that solved the problem!

closed 2023-05-08 by opc241