adelolmo/hd-idle #101: HDD in external bay not detected anymore

balouken, 2023-07-27:
Hello,

I've tested hd-idle yesterday.
The HDD did not spin down, so i pulled out the USB connector before i get to sleep.
But now i can't let the HDD spin up again?
Also drive is not detected (because not spinning up).
Is there a way to send a command to a drive that is not detected?

GrayXu, 2023-07-28:
I guess that the non-detected devices are likely caused by issues with the operating system rather than hd-idle. 

And for USB disks, you can try using the `-c ata` option.

balouken, 2023-07-28:
I don't know why this is happening.
But I was using hd-idle because HDD's didn't spin down. But now they don't spin up. Even when I use other external usb hub for the same HDD. Even on windows pc they don't spinning 🤷‍♂️.

I've searched -C ata, but can't find information about that. What does it do?

closed 2023-09-21 by adelolmo