adelolmo/hd-idle #104: Help understanding how to use the -p power_condition flag for SAS drives

weilah, 2023-09-06:
I'm trying to spin down and up of four **3TB HITACHI DKR2D-H3R0SS** hard disk drives controlled by a **PCIe Dell perc h200i card**.

I'm able to spin them down using: `HD_IDLE_OPTS="-i 3000 -l /var/log/hd-idle.log”` but as the documentation says they never come back again because "A stopped SAS disk will not start up automatically on access, but requires a startup command for reactivation"

I tried with `-p 2` and `-p 3` but none of them seem to do the trick. In fact when using either of these they never spin down.

Could someone please help me to undestand how to use the -p power_condition option? 
Thanks!

weilah, 2023-09-06:
On it! I'll try to do everything you suggested. Thanks

weilah, 2023-09-06:
So, I tried the following but to be honest I'm not sure what am I exactly looking for.
I ran `smartctl -a /dev/sda`
and got
```
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-6.1.0-0.deb11.9-amd64] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               HITACHI
Product:              DKR2D-H3R0SS
Revision:             FEFE
Compliance:           SPC-4
User Capacity:        3,000,592,982,016 bytes [3.00 TB]
Logical block size:   512 bytes
Rotation Rate:        7200 rpm
Form Factor:          3.5 inches
Logical Unit id:      0x5000cca03e9c37dc
Serial number:        XXXXXXX (I guess it wouldn't be a problem to post it.. but better save than sorry)
Device type:          disk
Transport protocol:   SAS (SPL-3)
Local Time is:        Wed Sep  6 12:24:24 2023 CEST
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Disabled or Not Supported

=== START OF READ SMART DATA SECTION ===
SMART Health Status: OK

Current Drive Temperature:     39 C
Drive Trip Temperature:        85 C
```

and other info which I think it isn't relevant Accumulated power on time..Vendor (Seagate Cache) information..SMART Self-test log.. but I can post it if needed

I also tried hdparm again not sure what I should aim for... I took a wild guess but didn't got back much.
Running `hdparm -B /dev/sda`
I got:
```
/dev/sda:
SG_IO: bad/missing sense data, sb[]:  72 05 20 00 00 00 00 34 00 0a 00 00 00 00 00 00 00 00 00 00 01 0a 00 00 00 00 00 00 00 00 00 00
```

and running `hdparm -C /dev/sda`
```
/dev/sda:
SG_IO: bad/missing sense data, sb[]:  72 05 20 00 00 00 00 34 00 0a 00 00 00 00 00 00 00 00 00 00 01 0a 00 00 00 00 00 00 00 00 00 00
SG_IO: bad/missing sense data, sb[]:  72 05 20 00 00 00 00 34 00 0a 00 00 00 00 00 00 00 00 00 00 01 0a 00 00 00 00 00 00 00 00 00 00
 drive state is:  unknown
```

a bit of guidance would be highly appreciated :)
Thanks in advance for your time

weilah, 2023-09-06:
I'm not able to spin them up after spinning them down. That's why I started to look closer at the documentation and the available options. 

I guess trying to `cd` any of their directories should be enough to wake them up, shouldn't it?

I took a look at the source code of hd-idle trying to understand how the `power_condition` flag be used... but I left clueless.

GrayXu, 2023-09-08:
@weilah `sg_start -s /dev/sdX` may be able to spin up SAS disks again.  
Back to the main problem, some vendors may only partially implement `power_condition` flag

weilah, 2023-09-08:
hey @GrayXu that worked! I was able to spin up the disk. That's progress some good progress. Thanks!

Can I somehow combine this with hd-idle? configuring it to run this command for the specified disk?
does this make sense what I'm saying?

GrayXu, 2023-09-11:
@weilah I doubt it🤔. Technically, hd-idle only monitors `/proc/diskstats`. Since READ doesn't access disks, hd-idle is unable to retrieve those records.

weilah, 2023-09-11:
Damn.. so close, yet so far. I'm clueless on how to move forward. Maybe I can try to find out whether my Hitachi drives implement the `power_condition`. 

I still haven't figured out how to use the flag. The documentation mentions two different values for this kind of disks: "Useful values for SAS disks are 2 for idle and 3 for standby." Should I choose 2 or 3? I'm unsure because they seem opposite to me, but I might be wrong.

Anyway, thanks for your help @GrayXu , really appreciate it

weilah, 2023-09-12:
Since the drives I want to spin down are only going to be used as a backup, I will probably end up setting a cron job to issue the command @GrayXu gave me shortly before the regular scheduled tasks to asure the drives are up and running

adelolmo, 2023-09-17:
@weilah 
I think you got to the right conclusion yourself already:
> Since the drives I want to spin down are only going to be used as a backup, I will probably end up setting a cron job to issue the command @GrayXu gave me shortly before the regular scheduled tasks to assure the drives are up and running.

hd-idle spins down disks. Technically it's able to do so by monitoring `/proc/diskstats`, as @GrayXu mentioned already.
But more importantly is that hd-idle does not spin up disks.

weilah, 2023-09-20:
Thank you both your support, very much appreciated. I will do as planned and spin them up manually. 

Still, I don't quite yet understand how the `power_condition` would be used. Which would be the right values to try, @adelolmo?

adelolmo, 2023-09-21:
@weilah 
There's a [comment in the code](https://github.com/adelolmo/hd-idle/blame/master/sgio/scsi.go#L34) added by @benengele (author of the feature) that will point you to the [Seagate documentation](https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf). In the section _3.49 START STOP UNIT command_ you'll find all details about the power condition.
Hope it helps.

weilah, 2023-09-21:
perfect, thanks @adelolmo. I'll close the thread, I think there is enough information about the topic.
And kudos on the hd-idle tool, very helpful.

closed 2023-09-21 by weilah