adelolmo/hd-idle #105: WD Re HDD over USB does not spin down.

amo13, 2023-10-12:
My Western Digital Re Disk attached over USB to my Raspberry Pi 2 B does not spin down. Neither when configured with idleTime=5:
```
disk=sda command=ata spunDown=false reads=32 writes=0 idleTime=5 idleDuration=4 spindown=0001-01-01T00:00:00 spinup=2023-10-12T21:15:23 lastIO=2023-10-12T21:15:23
disk=sda command=ata spunDown=false reads=32 writes=0 idleTime=5 idleDuration=5 spindown=0001-01-01T00:00:00 spinup=2023-10-12T21:15:23 lastIO=2023-10-12T21:15:23
sda spindown
disk=sda command=ata spunDown=true reads=32 writes=0 idleTime=5 idleDuration=5 spindown=2023-10-12T21:15:28 spinup=2023-10-12T21:15:23 lastIO=2023-10-12T21:15:23
disk=sda command=ata spunDown=true reads=32 writes=0 idleTime=5 idleDuration=6 spindown=2023-10-12T21:15:28 spinup=2023-10-12T21:15:23 lastIO=2023-10-12T21:15:23
```
...nor with `# hd-idle -t /dev/sda` or `# hd-idle -t /dev/sda -c ata` (both return nothing and nothing happens). 

The HDD *does* spin down though, if I do `# hdparm -y /dev/sda` or `# hdparm -Y /dev/sda`.

adelolmo, 2023-10-13:
@amo13 
Are you by any chance writing the hd-idle log into /dev/sda ?

amo13, 2023-10-14:
Thank you for responding. I appreciate your thoughts on this.

I don't think so. I used the systems unit and copied the log lines from the systemd journal.
The device itself is a luks volume. No partitions, just the full device as luks volume (with btrfs inside). But it was not decrypted or mounted when I tried to spin it down.

Anything else I can double-check or try?

adelolmo, 2023-10-14:
No partitions? That sounds familiar.
I think that's a know issue/limitation https://github.com/adelolmo/hd-idle/issues/70#issuecomment-1488037450
Could you please share your `cat /proc/diskstats`?

amo13, 2023-10-15:
I don't really understand the discussion in #70... Is there no workaround? 
If `hdparm` is able to spin down the disk, why wouldn't `hd-idle` be able to do the same?

Here is the result of diskstats:
```
$ cat /proc/diskstats
   1       0 ram0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       1 ram1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       2 ram2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       3 ram3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       4 ram4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       5 ram5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       6 ram6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       7 ram7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       8 ram8 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1       9 ram9 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      10 ram10 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      11 ram11 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      12 ram12 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      13 ram13 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      14 ram14 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   1      15 ram15 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       0 loop0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       1 loop1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       2 loop2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       3 loop3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       4 loop4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       5 loop5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       6 loop6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       7 loop7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 179       0 mmcblk0 2710 1733 178235 15005 264 353 11177 37596 0 0 52602 0 0 0 0 0 0
 179       1 mmcblk0p1 93 0 4193 325 1 0 1 2 0 0 327 0 0 0 0 0 0
 179       2 mmcblk0p2 2543 1733 171658 14559 263 353 11176 37594 0 0 52153 0 0 0 0 0 0
   8       0 sda 4 0 32 20 0 0 0 0 0 0 20 0 0 0 0 0 0
```

Thanks for your help with this!

adelolmo, 2023-10-20:
I think that, given what you shared, I'm asking the wrong questions.
To be honest, I can't tell what's going on. The program seems to behave as expected, as shown in the log, but the spin down commands just don't work.

amo13, 2023-10-20:
I'm wondering too. It looks odd to me that `hdparm -y /dev/sda` *does* work though...
Anything we can do or check or try?

adelolmo, 2023-10-20:
Could you share the result of running `# hd-idle -t /dev/sda` and `# hd-idle -t /dev/sda -c ata` with the flag `--verbose`?

amo13, 2023-10-20:
The result is absolutely nothing:

```
[alarm@alarmpi ~]$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    0  1.6T  0 disk 
mmcblk0     179:0    0  3.7G  0 disk 
|-mmcblk0p1 179:1    0  200M  0 part /boot
`-mmcblk0p2 179:2    0  3.5G  0 part /
[alarm@alarmpi ~]$ sudo hd-idle -t /dev/sda --verbose
[sudo] password for alarm: 
[alarm@alarmpi ~]$ sudo hd-idle -t /dev/sda -c ata --verbose
[alarm@alarmpi ~]$ 
```

adelolmo, 2023-10-21:
> The result is absolutely nothing:

Of course, that's because I gave you the wrong commands 🤦‍♂️

Please try again:

- `# hdparm --verbose -y /dev/sda`
- `# hdparm --verbose -Y /dev/sda`

amo13, 2023-10-21:
I see :)
Here we go again:
```
[alarm@alarmpi ~]$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    0  1.6T  0 disk 
mmcblk0     179:0    0  3.7G  0 disk 
|-mmcblk0p1 179:1    0  200M  0 part /boot
`-mmcblk0p2 179:2    0  3.5G  0 part /
[alarm@alarmpi ~]$ sudo hdparm --verbose -y /dev/sda
[sudo] password for alarm: 

/dev/sda:
APT: USB ID = 0x152d:0x2339 (0x100)
APT: Found supported device jmicron
outgoing cdb:  df 10 00 00 01 00 72 0f 00 00 00 fd
SG_IO: ATA_12 status=0x0, host_status=0x0, driver_status=0x0
APT: JMicron Port: 0xA0
 issuing standby command
outgoing cdb:  df 10 00 00 00 00 00 00 00 00 a0 e0
[alarm@alarmpi ~]$ sudo dd if=/dev/sda bs=4096 count=1 of=/dev/null iflag=direct
1+0 records in
1+0 records out
4096 bytes (4.1 kB, 4.0 KiB) copied, 12.2197 s, 0.3 kB/s
[alarm@alarmpi ~]$ sudo hdparm --verbose -Y /dev/sda

/dev/sda:
APT: USB ID = 0x152d:0x2339 (0x100)
APT: Found supported device jmicron
outgoing cdb:  df 10 00 00 01 00 72 0f 00 00 00 fd
SG_IO: ATA_12 status=0x0, host_status=0x0, driver_status=0x0
APT: JMicron Port: 0xA0
 issuing sleep command
outgoing cdb:  df 10 00 00 00 00 00 00 00 00 a0 e6
SG_IO: ATA_12 status=0x0, host_status=0x0, driver_status=0x0
[alarm@alarmpi ~]$
```

adelolmo, 2023-10-21:
The issue is that hd-idle always sends the SG_ATA with length 16, but in your case hdparm is using SG_ATA_12.
I will treat it as a bug in hd-idle and try to come up with a patch to support ATA_12 too.

amo13, 2023-10-21:
Alright, thank you. Looking forward to testing it!
Have you got paypal so I can buy you a coffee?

adelolmo, 2023-10-22:
I've created a pre-release that might solve your issue: https://github.com/adelolmo/hd-idle/releases/tag/v1.21
You should try this command instead: `# hd-idle -t /dev/sda -c ata12`

I still haven't figured out how hdparm decides whether using ata16 or ata12 ¯\_(ツ)_/¯
My idea is to have you trying the pre-release first of all, and if it works I'll spend some more time thinking about how to have hd-idle deciding between ata16 or ata12. I'd like to avoid users having to make that decision if possible.

> Have you got paypal so I can buy you a coffee?

I appreciate the intention, but I prefer not to get any money involved on this.

amo13, 2023-10-22:
Thank you very much for your efforts on this!

Unfortunately, nothing happens with the new command either. Is there anything I can do to help making it work?

If that is of any help, here are some SMART info about the disk:
<details>
<summary># smartctl --info /dev/sda</summary>

```
[alarm@alarmpi build]$ sudo smartctl --info /dev/sda
smartctl 7.4 2023-08-01 r5530 [armv7l-linux-6.1.57-2-rpi-ARCH] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Western Digital Re
Device Model:     WDC WD4000FYYZ-01UL1B2
Serial Number:    WD-WMC130E8PHYT
LU WWN Device Id: 5 0014ee 6051f68d0
Firmware Version: 01.01K03
User Capacity:    4,000,787,030,016 bytes [4.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Device is:        In smartctl database 7.3/5528
ATA Version is:   ATA8-ACS (minor revision not indicated)
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 1.5 Gb/s)
Local Time is:    Sun Oct 22 08:38:40 2023 UTC
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```

</details>

<details>
<summary># smartctl --identify /dev/sda</summary>

```
[alarm@alarmpi build]$ sudo smartctl --identify /dev/sda
smartctl 7.4 2023-08-01 r5530 [armv7l-linux-6.1.57-2-rpi-ARCH] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== ATA IDENTIFY DATA ===
Word     Bit     Value   Description

   0      -     0x427a   General configuration
   0     14:8     0x42   Vendor specific [RET-3]
   0      6          1   Not removable controller and/or device [OBS-6]
   0      5:3      0x7   Vendor specific [RET-3]
   0      1          1   Vendor specific [RET-3]

   1      -     0x3fff   Cylinders [OBS-6]

   2      -     0xc837   Specific configuration (0x37c8/738c/8c73/c837)

   3      -     0x0010   Heads [OBS-6]

   6      -     0x003f   Sectors per track [OBS-6]

  10-19   -     .        Serial number (String)
  10-13   .     0x2020:2020:2057:442d  "     WD-"
  14-17   .     0x574d:4331:3330:4538  "WMC130E8"
  18-19   .     0x5048:5954            "PHYT"

  23-26   -     .        Firmware revision (String)
  23-26   .     0x3031:2e30:314b:3033  "01.01K03"

  27-46   -     .        Model number (String)
  27-30   .     0x5744:4320:5744:3430  "WDC WD40"
  31-34   .     0x3030:4659:595a:2d30  "00FYYZ-0"
  35-38   .     0x3155:4c31:4232:2020  "1UL1B2  "
  39-42   .     0x2020:2020:2020:2020  "        "
[... 312 lines cut ...]
200 Multi_Zone_Error_Rate   0x0008   200   200   000    Old_age   Offline      -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Conveyance offline  Completed without error       00%     26549         -
# 2  Short offline       Completed without error       00%     26549         -

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.

The above only provides legacy SMART information - try 'smartctl -x' for more
```

</details>

adelolmo, 2023-10-24:
I had a second look at the output of the command `sudo hdparm --verbose -y /dev/sda`.
There seem to be two ata commands send:
1. `df 10 00 00 01 00 72 0f 00 00 00 fd`
1. `df 10 00 00 00 00 00 00 00 00 a0 e0`

The second one is the standby command, but the first one looks like something specific for your device (jmicron).
I backed new pre-release packages with a version that has these commands hardcoded with the commands above. 
Could you please give it a try and let me know if it works? Just run `# hd-idle -t /dev/sda -c ata12`

amo13, 2023-10-26:
Amazing! I can confirm that it does work now. And it outputs
```
outgoing cdb:  df 10 00 00 01 00 72 0f 00 00 00 fd 
 issuing standby command
outgoing cdb:  df 10 00 00 00 00 00 00 00 00 a0 e0
```
I guess that adding some detection logic is not quite easy.. If you'd like me to test something else, I'll be glad to help.

adelolmo, 2023-10-28:
After some investigation I think I came up with a way to identify if the jmicron controller is used. `hdparm` has special handling for that controller and now `hd-idle` should too.
Could you please test again this version (packages are re-uploaded): https://github.com/adelolmo/hd-idle/releases/tag/v1.21
You should try: `# hd-idle -d -t /dev/sda -c ata`

amo13, 2023-10-29:
Just tested. Unfortunately, it does not work. The output is:
```
 issuing standby command
outgoing cdb:  85 06 00 00 00 00 00 00 00 00 00 00 00 40 e0 00
```

adelolmo, 2023-10-29:
Thanks for testing it.
I changed the part that identifies the device based on the vendor and product. Hopefully this time works.
Could you please give it another try?

amo13, 2023-10-29:
Thank you too!
We're not quite there yet:
`# hd-idle -d -t /dev/sda -c ata`
```
APT: USB ID = 0x152d:0x2339 (0x0100)
APT: Unsupported device
 issuing standby command
outgoing cdb:  85 06 00 00 00 00 00 00 00 00 00 00 00 40 e0 00
```

I just cloned the repo and compiled with `make`.

adelolmo, 2023-10-29:
> I just cloned the repo and compiled with `make`.

Well, that's way faster :)

One more attempt. You can pull, compile the latest changes and try again.

If it doesn't work this time, could you also share the output of these commands:
* `# cat /sys/block/sda/../../../../../../idVendor`
* `# cat /sys/block/sda/../../../../../../idProduct`
* `# cat /sys/block/sda/../../../../../../bcdDevice`

If it doesn't find the file you might have to change the relative path:
eg. `# cat /sys/block/sda/../../../../../idVendor` (with 5 path jumps)
eg. `# cat /sys/block/sda/../../../../../../../idVendor` (with 7 path jumps)

amo13, 2023-10-29:
Fantastic, it works now! 
Thank you

adelolmo, 2023-10-29:
This one was fun to get working, but I'm glad it's now done :sweat_smile: 
Closing now.

closed 2023-10-29 by adelolmo