adelolmo/hd-idle #106: Can't spindown my internal WD Blue HDD

thordal-de, 2023-10-23:
I have two internal SATA HDD disks and would like to spindown the second one /dev/sdb (WD blue 2 TB, WD20EZBX) since there is MS Windows on it. On the first one /dev/sda runs Linux Mint "Victoria".

I managed to set the /dev/sdb into the standby- modus, but I was not able to spin it down. For this task I use hd-idle, which I installed successfully using the guide on github [ https://github.com/adelolmo/hd-idle](url).
`/etc/default/hd-idle`
`HD_IDLE_OPTS="-h"
START_HD_IDLE=true
HD_IDLE_OPTS="-i 0 -a sdb -i 300 -a sda -i 1800"`
According to the kernel and syslog, hd-idle should run:
`$ ps aux | grep hd-idle

root        1140  0.0  0.0   2776   896 ?        S    10:54   0:00 /usr/sbin/hd-idle -i 0 -a sdb -i 300 -a sda -i 1800`
`$ sudo cat /var/log/syslog | grep hd-idle

Oct 23 10:54:08 tux systemd[1]: Stopping hd-idle - spin down idle hard disks...
Oct 23 10:54:34 tux systemd[1]: Starting hd-idle - spin down idle hard disks...
Oct 23 10:54:34 tux systemd[1]: Started hd-idle - spin down idle hard disks.`
After about 5 minutes the sdb stanbies:
`$ sudo hdparm -C /dev/sdb

drive state is:  standby`

BUT /dev/sdb spins further, there's no spin down as I want it to. 
`sudo hdparm -y /dev/sdb` results in the same drive stats, but spin down neither. Maybe it is becaus APM seems not to be supported
`$ sudo hdparm -B /dev/sdb

/dev/sdb:
 APM_level	= not supported`
but other Forum recommend to post my issue here since hd-idle should be experimental.

How can I solve my issue? 
Thanks

adelolmo, 2023-10-23:
The configuration in `/etc/default/hd-idle` shouldn't contain two instances of `HD_IDLE_OPTS`.
This is what you need:
> START_HD_IDLE=true 
HD_IDLE_OPTS="-i 0 -a sdb -i 300 -a sda -i 1800"

Could you please try the following commands and share here the output?
* `sudo hdparm --verbose -y /dev/sdb`
* `hd-idle -t /dev/sdb -c scsi`
* `hd-idle -t /dev/sdb -c ata`

BTW. What do you mean by hd-idle been _experimental_?

thordal-de, 2023-10-23:
> sudo hdparm --verbose -y /den/sdb:

`dev/sdb:
APT: No idVendor found -> not USB bridge device
 issuing standby command
outgoing cdb:  85 06 20 00 00 00 00 00 00 00 00 00 00 40 e0 00
SG_IO: ATA_16 status=0x2, host_status=0x0, driver_status=0x8
SG_IO: sb[]:  72 01 00 1d 00 00 00 0e 09 0c 00 00 00 00 00 00 00 00 00 00 40 50 00 00 00 00 00 00 00 00 00 00
SG_IO: desc[]:  09 0c 00 00 00 00 00 00 00 00 00 00 40 50
      ATA_16 stat=50 err=00 nsect=00 lbal=00 lbam=00 lbah=00 dev=40
`

> hd-idle -t /dev/sdb -c scsi: and hd-idle -t /dev/sdb -c ata

`/dev//dev/sdb: No such file or directory
`
Thart hd-idle should be experimental was an answer from the Linux Mint forum:

> hd-idle is experimental software sometimes it works, sometimes not, see [viewtopic.php?t=387066](https://forums.linuxmint.com/viewtopic.php?t=387066) and many posts on other forums

Thanks

adelolmo, 2023-10-23:
The output of the command `hd-idle -t /dev/sdb -c ata` seems weird: `/dev//dev/sdb: No such file or directory`
It looks like you have the original version of hd-idle installed.
Could you please make sure that you have followed [this steps](https://github.com/adelolmo/hd-idle#install) to install hd-idle?

thordal-de, 2023-10-24:
OK, now I've installed the hd-idle_1.20_amd64.deb package and have activated it without any incident.
The 
$ cat /var/log/syslog | grep hd-idle 
looks a bit different now
`Oct 24 07:44:34 tux systemd[1]: Started hd-idle - spin down idle hard disks.
Oct 24 07:44:34 tux hd-idle[4228]: symlinkPolicy=0, defaultIdle=0, defaultCommand=scsi, defaultPowerCondition=0, debug=false, logFile=, devices={name=sda, givenName=sda, idle=1800, commandType=scsi, powerCondition=0}{name=sdb, givenName=sdb, idle=300, commandType=scsi, powerCondition=0}
Oct 24 07:44:54 tux systemd[1]: Stopping hd-idle - spin down idle hard disks...
Oct 24 07:45:20 tux systemd[1]: Started hd-idle - spin down idle hard disks.
Oct 24 07:45:20 tux hd-idle[1139]: symlinkPolicy=0, defaultIdle=0, defaultCommand=scsi, defaultPowerCondition=0, debug=false, logFile=, devices={name=sda, givenName=sda, idle=1800, commandType=scsi, powerCondition=0}{name=sdb, givenName=sdb, idle=300, commandType=scsi, powerCondition=0}`

$ sudo hdparm --verbose -y /dev/sdb
`/dev/sdb:
APT: No idVendor found -> not USB bridge device
 issuing standby command
outgoing cdb:  85 06 20 00 00 00 00 00 00 00 00 00 00 40 e0 00
SG_IO: ATA_16 status=0x2, host_status=0x0, driver_status=0x8
SG_IO: sb[]:  72 01 00 1d 00 00 00 0e 09 0c 00 00 00 00 00 00 00 00 00 00 40 50 00 00 00 00 00 00 00 00 00 00
SG_IO: desc[]:  09 0c 00 00 00 00 00 00 00 00 00 00 40 50
      ATA_16 stat=50 err=00 nsect=00 lbal=00 lbam=00 lbah=00 dev=40`

Both
$ sudo hd-idle -t /dev/sdb -c scsi
$ hd-idle -t /dev/sdb -c ata
result in "nothing". No output an no spin down.

adelolmo, 2023-10-24:
> After 300 seconds /dev/sdb switches to standby, (sudo hdparm -C /dev/sdb => standby) but spins further.

This means that the disk is spun down (standby). Why it spins up afterwards is something I can't help you with.

> BTW: I wonder why the newest version of hd-idle is not available by using the Advanced Package Tool (apt) by Linux Mint?

Because no one has taken the time and effort to make it available there.

closed 2023-10-24 by adelolmo