adelolmo/hd-idle #107: HD-Idle command does not seem to get set?

remirr, 2023-10-24:
Hi there,

I am trying to set up a new HD-Idle command and log because last night I noticed my disk was spinning up and down every 2-3 minutes and I have no idea why. I have now tried to set it to 30 minutes with a log file but when I press enter it never seems to save the config, any idea why?

Also, I really only need this disk spun up at 5pm and then spun down at midnight each day and otherwise it can be off, is there a way to do this?

Thank you
[https://ibb.co/stFWvSC](url)

adelolmo, 2023-10-24:
Your configuration says that `sda` should spin down after 1800s of inactivity (30mins).
I don't understand what `when I press enter it never seems to save the config` means.

> I really only need this disk spun up at 5pm and then spun down at midnight each day and otherwise it can be off, is there a way to do this?

You already have it configured to spin down after 30mins of inactivity.

rabelux, 2023-10-24:
@remirr you know about config files, right?
Spinning up should happen automatically on drive access and spinning down at 12pm can be done via cron job.

remirr, 2023-10-24:
So as you can see from the image, I tried setting the config for 30 mins, I think it is 10 now and when I hit enter it just sits there like that. It’s like it never saves the command because when I hit ctrl-c it cancels and goes back to waiting for a new command….

Also I tried getting the log to work as you can see in the image but nothing is showing up

so that’s the problem I am having, for some reason the disc keeps spinning up and down every few minutes but there should be nothing accessing it, obviously this isn’t working so I was wondering if there was a way to force it off and then force it back on at certain times.

adelolmo, 2023-10-25:
I see where your confusion comes from.
Please read the following section of the README:
* https://github.com/adelolmo/hd-idle#Configuration
* https://github.com/adelolmo/hd-idle#run-hd-idle

closed 2023-10-25 by adelolmo