adelolmo/hd-idle #108: Drive standby instead of sleep.

olokelo, 2023-11-04:
Hi,

First of all, it's really a great project. I was forced to explore something different than `hdparm` because it wouldn't automatically spin down drives in my newest setup.

I find `hd-idle` easier to configure and the ability to see the logs when drives where woken up or spun down is very nice addition.

I'm currently running a systemd service with the following options:
```
HD_IDLE_OPTS="-i 0 -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4JT1V3B -i 900 -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4JXSX9B -i 900 -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4KA0DTB -i 900 -a /dev/disk/by-id/ata-WDC_WD40PURX-64GVNY0_WD-WCC4E3SH4ZPY -i 900 -a /dev/disk/by-id/ata-WDC_WD40PURX-64GVNY0_WD-WCC4E6PSKY57 -i 900 -l /var/log/hd-idle.log"
```

Drives are put into sleep as expected but I wonder if there is a way to put them into standby instead?
Current behavior is similar issuing `hdparm -Y /dev/drive`. When I manually run `hdparm -y /dev/drive` it also works and I feel drives wake up a bit faster. Also I'm able to list directories even when the drives are now spinning (kind of magic to me but there's probably a good explanation of how it works).

It doesn't work like that when drives were put into sleep by `hd-idle` but I wonder if there's maybe an option for `hd-idle` to put drives into standby instead of completely shutting them down.

adelolmo, 2023-11-04:
You could use the `ata` command instead. The `ata` command will set your disks to standby the same way `hdparm` does.

```
HD_IDLE_OPTS="-i 0 -c ata -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4JT1V3B -i 900 -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4JXSX9B -i 900 -a /dev/disk/by-id/ata-HGST_HDN726040ALE614_K4KA0DTB -i 900 -a /dev/disk/by-id/ata-WDC_WD40PURX-64GVNY0_WD-WCC4E3SH4ZPY -i 900 -a /dev/disk/by-id/ata-WDC_WD40PURX-64GVNY0_WD-WCC4E6PSKY57 -i 900 -l /var/log/hd-idle.log"
```

olokelo, 2023-11-04:
Thank you, now works as expected! :)

```bash
$ hdparm -C /dev/sd{d,e,g,h,i}

/dev/sdd:
 drive state is:  standby

/dev/sde:
 drive state is:  standby

/dev/sdg:
 drive state is:  standby

/dev/sdh:
 drive state is:  standby

/dev/sdi:
 drive state is:  standby
```

closed 2023-11-04 by olokelo