adelolmo/hd-idle #109: Logfile won't be generate data。

Yxini, 2023-11-28:
hi！
I modified the "/etc/default/hd idle" file.
`nano /etc/default/hd-idle`
The following content has been modified:
`START_HD_IDLE=true`
`HD_IDLE_OPTS="-i 0 -c ata -a /dev/disk/by-id/ata-TOSHIBA_DT01ACA100_55746MGRS -i 300 -l /var/log/hd-idle.log"`
Then, I restarted the service.
`systemctl restart hd-idle.service`
and then i waited for five minutes and saw that the hard drive was sleeping normally. But it was found that there was no data in the log file.

[image]

[image]

Yxini, 2023-11-29:
oh，thankyou。
The next day, I saw that the file had generated data.
It seems that I misunderstood the way the file was generated.

Yxini, 2023-11-29:
I have another question, which is that I have a device that uses USB to convert to Sata. Due to this particularity, I want the hard drive to not be completely dormant and reduce the speed (similar to Seagate's idle_c). May I ask if there is a way to do it.

adelolmo, 2023-11-29:
Speed reduction is not supported.

Yxini, 2023-11-29:
Okay, thank you for your reply！

closed 2023-11-29 by Yxini