adelolmo/hd-idle #110: Hook-function to be called after spinning down

squonk11, 2023-12-07:
I would like to propose an additional feature:
Implementation of a hook-function (e.g. call of external shell script, if present) after the disks successfully spun down. With this functionality I could be able switch off the HD-Bay which (in my case) makes a terrible noise from the fans and still has a power consumption of 10W even if the disks are spun down. 
But there also needs to be a hook-function for spinning up the disks when accessing them again - is this also handled by hd-idle or is this an OS-function?

adelolmo, 2023-12-07:
This is a topic that already came up in the past: https://github.com/adelolmo/hd-idle/issues/49

squonk11, 2023-12-19:
I solved my requirement by using std on-board tools: cron and autofs. 
For those who are interested in the solution:
I am using a shell-script via a cron-job which checks the result of hdparm in order to power-off the drive bay like so:
```
#!/usr/bin/env bash
if [[ -n $( /usr/sbin/hdparm -C /dev/disk/by-uuid/05628b67-de44-44f1-a3fe-2cd4e0cbaf26 2>&1 | grep "standby" ) ]]; then
  if [[ -n $( /usr/sbin/hdparm -C /dev/disk/by-uuid/40322f73-89ce-423d-a29d-14157a412007 2>&1 | grep "standby" ) ]]; then
    /usr/bin/curl "http://192.168.1.76/relay/0?turn=off"
    echo -e "\nHDs switched off at: $( date )"
  fi
fi
```
For powering the drives up again I used autofs by setting up an auto.nas file:
```
#!/bin/bash
if [[ -n $( /usr/bin/curl -s http://192.168.1.76/relay/0 | grep '"ison":false' ) ]]; then
   /usr/bin/curl "http://192.168.1.76/relay/0?turn=on" > /dev/null 2>&1
   /usr/bin/sleep 20 > /dev/null 2>&1
fi
case $1 in
        nas1) echo "-fstype=ext4,rw :/dev/disk/by-uuid/05628b67-de44-44f1-a3fe-2cd4e0cbaf26"
              ;;
        nas2) echo "-fstype=ext4,rw :/dev/disk/by-uuid/40322f73-89ce-423d-a29d-14157a412007"
              ;;
esac
```
For switching the power on and off I am using a Shelly Plug S.

Maybe this info might help someone.

closed 2023-12-19 by adelolmo