adelolmo/hd-idle #111: Spin up not recognized

fidoriel, 2023-12-27:
Hey,

with my new Seagate Exos I have got some issues.
Manual spin down works.
hd-idle recognizes disk.
Now the strange part begins: hd-idle does spin down the disk via scsi and ata. If I wait for spin down, the disk spins down and hd idle shows spunDown=true.
So far, so good. If I get back after some time, the disk was not used, same lastIO. The disk is spinning, but hd idle does show it is spunDown and the lastIO is some hours ago. I do not why the disk spins up, smart is disabled.

Thomas-8-Bit, 2024-02-14:
Hey,

same here. The first spindown after starting hd-idle works as aspected. But if the harddisk (usb-harddisk) is spinup again, hd-idle doesn't recognize it and so doesn't spindown again.

fidoriel, 2024-02-14:
Oh, I see this issue is still open: Plot twist, smart was not disabled. A docker container had a smart demon running.

closed 2024-02-14 by fidoriel