adelolmo/hd-idle #118: Immediate spin up after issuing spin down command

TWry, 2024-09-22:
When I try to spin down my external USB drive it immediately (physically) spins up again. I am guessing that this may be caused by a NFS share that I have created for this disk but I am not sure. How can I debug what is causing this behaviour?

The drive is a Seagate RSS LLC Backup Plus Hub 8TB and the driver identifies as "uas". Latest hd-idle version is installed.

darrell-k, 2024-11-25:
Did you just format/create a new Ext4 file system? The defaults leave a lot of work to be done when it is first mounted (specifically writing inode tables) which is done by the kernel, very slowly so as not to interfere to much with regular disk access. This means a big disk could be busy for hours or more on first mount. 

I just had the same issue with a new 8TB disk, and although I'd come across the issue before, I'd completely forgotten about it. Finally remembered! I decided to recreate the file system with the options to do this work up front, meaning that took 10 or 15 minutes, but straight afterwards hd-idle worked as expected.

You can also force it with a one-off mount option if you've already lots of data on the disk so don't want to recreate the file system, or just leave it plugged in and wait. All explained here: See https://fedetft.wordpress.com/2022/01/23/on-ext4-and-forcing-the-completion-of-lazy-initialization/

adelolmo, 2024-11-25:
You could also have a look to `fatrace`, as mentioned in [this comment](https://github.com/adelolmo/hd-idle/issues/120#issuecomment-2466273055).