adelolmo/hd-idle #120: Unexpected spin up of snapraid parity disk

zz0rk, 2024-11-08:
Behavior: hd-idle unexpectedly spun up parity disk on 11/08 although disk has not been accessed since 11/04. Snapraid did not initiate a sync and the parity disk is not being shared (NFS or SMB) or written to by any other process.

Expected behavior: hd-idle should not spin up parity disk unless it's being accessed.

Thank you

Debian 12
hd-idle version: 1.21 / 2023-10-22
snapraid version: v12.3

Log:
/var/log# cat hd-idle.log 
date: 2024-11-08, time: 07:37:50, disk: /dev/disk/by-id/ata-HGST_HUH721010ALE604_2TJE7KJD, running: 3601, stopped: 51723
date: 2024-11-08, time: 11:08:53, disk: /dev/disk/by-id/ata-HGST_HUH721010ALE604_2TJE7KJD, running: 3600, stopped: 9063

Parity drive:
ls -latr /mnt/raid/disk-00-parity/
total 8568933508
drwx------ 2 root root          16384 Mar 31  2024 lost+found
drwxr-xr-x 3 root root           4096 Mar 31  2024 .
drwxrwxr-x 7 root users          4096 Apr 16  2024 ..
-rw------- 1 root root  8774585155584 Nov  4 15:52 snapraid.parity

zz0rk, 2024-11-09:
hd-idle:
/etc/default/hd-idle
	START_HD_IDLE=true
	# parity
	HD_IDLE_OPTS="-i 0 -c ata -s 0 -l /var/log/hd-idle.log -a /dev/disk/by-id/ata-HGST_HUH721010ALE604_2TJE7KJD -i 3600

adelolmo, 2024-11-09:
Hi @zz0rk 
> hd-idle unexpectedly spun up parity disk

Sorry, but hd-idle is not able to spin up disks, it only can spin them down.
You should extend the search to find the culprit.

closed 2024-11-09 by adelolmo

zz0rk, 2024-11-09:
Other folks who run into this issue may want to give 'fatrace' a shot.
https://github.com/martinpitt/fatrace

fatrace -t -c -o /var/log/fatrace

07:38:27.584347 find(3102642): O   /mnt/raid/disk-00-parity
07:38:27.584665 find(3102642): R   /mnt/raid/disk-00-parity
07:38:27.584712 find(3102642): C   /mnt/raid/disk-00-parity

The 'find' command was waking up my parity disk.